pylessard/python-udsoncan #1: Scapy with UDS and ISO-TP support

polybassa, 2018-04-17:
Hi @pylessard 
I just found your repos.
You did a very similar thing, like I do at the moment.
I'm trying to bring ISOTP, UDS and many more automotive protocols into Scapy mainline.

Would be great, if we could collaborate on this task in Scapy mainline.

Here is an example branch of my implementations:
https://github.com/polybassa/scapy-1/tree/isotp
It't not mainline, but I'm working with the maintainers on Scapy on bringing it there.

Greetings Nils

pylessard, 2018-04-17:
Hi Nils, 
Interresting, I've never heard of Scapy before. As you have certainly understood, this project handle UDS only, IsoTP is not included. Communication must go through a Socket.

I am open to the idea, but I will keep this repo running even if Scapy includes UDS (or at least, until I feel like Scapy has made this one obsolete). 

In the short term, I want to focus into releasing V1.0. We can look into this afterward. 

I added my email to my profile temporarily, provide me with some contact informations.

Regards

pylessard, 2018-04-17:
I closed this issue as it belongs to Scapy project. This issue was a way to communicate, working around the lack of private messaging on Github.

closed 2018-04-17 by pylessard