pylessard/python-udsoncan #2: use python-can as CAN layer?

mgiaco, 2018-07-02:
Hi,

Do you know python-can. I am wondering if it is possible to use that for the CAN layer.

https://python-can.readthedocs.io

cheers

pylessard, 2018-07-03:
Hi, 
The answer is No, but if you limit yourself the UDS messages smaller than 8 bytes, you could make it work.

There is 2 layers underneath the UDS protocol. CAN is the lowest level (the datalink layer). To send UDS messages, you need to send frame larger than the CAN 8 bytes capability. The IsoTP protocol is the transport layer between the CAN layer and the UDS software that concatenate the CAN message a create a 4095 long frame.   python-can gives you access to the CAN layer only.

If your messages are smaller than 8 bytes, you could use the QueueConnection and forward the 8 bytes frames to python-can or you could make your own Connection that does that.

Working with 8 bytes frame is a very big limit, but it could work if you read a single DTC at the time or if you know that the DID you read fits.

If you work under Linux, you should look at this repo : https://github.com/hartkopp/can-isotp
If you are under windows, I believe Peak offer a DLL that handle IsoTP, I never tried it.

Finally, you could look at PyVit, I believe they have a IsoTP layer that is implemented in Python (therefore in the user space).

Hope that helps

pylessard, 2018-08-24:
Awesome, I see that you are making a Python implementation of IsoTP (in user space). How well does this behave timing wise? 

I would expect variable result depending on the platform. I believe Windows has a thread resolution of 16ms. I'd be interested by your benchmark

Thank you!

christiansandberg, 2018-08-24:
The timing will probably not be great, but hopefully most ECUs will be able to handle it. I guess it is OK if the delay between messages is greater than what the server requests, but might need to take care of it if the delay gets too low.

acorn50, 2018-09-17:
Does this library support multi-frame messages?The first byte of the message defines the valid byte，not the DID.

pylessard, 2018-09-17:
@acorn50 You may be confusing few things here. I assume you are talking about the UDS package when you say "this library".  As you can see [in the doc](https://udsoncan.readthedocs.io/en/latest/udsoncan/connection.html), the UDS package doesn't care about the CAN layer, multi packet must be handled by the underlying layer.  IsoTP is one protocol that can do multi frame messages and @christiansandberg is presently working on implementing it in Python. It can already be done with the non-official [Linux driver](https://github.com/hartkopp/can-isotp). 

I am quite confused about "The first byte of the message defines the valid byte，not the DID."  This hardly make sense. DID are only relevant when using a service that transfer DID, which is not always the case. And I can't tell what you mean by "the valid byte".  

Hopefully, I answered your question with my first paragraph

acorn50, 2018-09-18:
for example,the message like 0x03,0x22,0xf1,0x80,0x00,0x00,0x00,0x00,the first byte 0x03 means that 0x22,0xf1,0x80 is valid bytes.It delimits the length of valid bytes.And the response message like 0x10,0x11,0x62,0xf1,0x80,0xff,0xff,0xff
0x21,0xff,0xff,0xff,0xff,0xff,0xff,0xff
0x22,0xff,0xff,0xff,0xff,0x00,0x00,0x00
the fisrt byte represent the sequence of message.
how can i process this case?

pylessard, 2018-09-18:
Hi, 
So you are talking about IsoTP. The first byte is not necessarely the length of the message. The first 4 bits tells the type of message. The length of the message will be encoded differently depending on the message type.   In your first case, 0x03 is message type 0 (Single Frame), then the length is encoded in the lower 4 bits (3).  In the response, you have a longer frame, bytes 0x10,0x11 means a message of type=1 (First Frame), length is encoded on 12 bits, therefore : 0x011. Following that, 0x21 is message of type=2 (Consecutive Frame), no length is encoded, it is only the continuation of the message.

See the below table. DL is "Data Length"

[image]

To handle these message, you need a library of code that handle the IsoTP protocol, and this is what we are talking here. Either use an existing library being from Linux or a commercial one or use a user space implementation, either PyVit or aisotp (work in progress). In any case, you should also learn the protocol, try to get your hand on the ISO-15765-2 standard.

This being said,  Github is not the best place to ask for technical help. We need to keep our conversation focused on the development of the projects for the sake of clarity.  I think stackoverflow.com would be more adequate if you have further questions.

acorn50, 2018-09-18:
thanks a lot,it's very useful.

AokiAhishatsu, 2018-10-05:
I'm on something similar at the moment.
Is there somewhere documented with which OS, CAN adapter and settings this library is developed/tested?

christiansandberg, 2018-10-05:
With python-can you would be able to use the library directly on Windows with a variety of CAN adapters. See [the docs](https://python-can.readthedocs.io/en/stable/interfaces.html).

I have been waiting for v3.0 to be release so now I can proceed with the implementation.

jutshanahan, 2018-10-23:
@christiansandberg .  Thanks for working on this.  Would like to test udsoncan with python-can when you release.

christiansandberg, 2018-10-23:
There is a branch in my fork that might work, see #6.

pylessard, 2019-01-02:
Release v1.7 supports python can. 
See documnetation, the is now a connection named PythonIsoTpConnection which requires the isotp module [here](https://github.com/pylessard/python-can-isotp) version > 1.3 (broken bind() interface with v0.x)

See [an example ](https://udsoncan.readthedocs.io/en/latest/udsoncan/examples.html#using-uds-over-python-can).  I would have liked to make it work with a one-liner, but had to make it a little bit more complex to support all addressing mode.

Cheers

closed 2019-01-02 by pylessard