pylessard/python-udsoncan #5: Problemi installing package from pip

fbagagli, 2018-08-03:
Hi,
first of all thanks for the huge work on this module, especially for the complete test suite that came with it.

Installing the module on a clean environment with python 3.7 on Ubunt 18.04, i've found that the module came installed without the udsoncan/services directory, and also the logging.conf file is missing.
So if you run "import udsoncan" from an interactive python session, an exception is raised

These are the steps to reproduce:

$ pipenv --python 3.7
$ pipenv shell
$ pip install udsoncan
$ python (to enter interactive session)
import udsoncan (<- this raise the exception)

I've modified the setup.py that seems to resolve this problem, and i've created this pull request ( #4 )

Thanks

pylessard, 2018-08-03:
Pull request merged. Thank you very much!

closed 2018-08-03 by pylessard

pylessard, 2018-08-03:
Included in release v1.3