pylessard/python-udsoncan #8: I have some confuse about communition type data type convert in communitcation control module

kazenoai, 2018-10-26:
[image]
[image]

It will execute from_byte if the communication_type is int type.  But it will judge if communication_type is bytes type in from_byte function.  Variable val in from_byte function does not be assigned a value if the communication_type is not a bytes type. It causes an error when I run client.communication_control function.

pylessard, 2018-10-26:
Can you share an example of code that is failing?

kazenoai, 2018-10-26:
I rewrite the tp layer by python-can. It is not fully complete now. Attachment test.zip include tp.py file and the connection.py modify by me. ForTest.py is the test code.

[test.zip](https://github.com/pylessard/python-udsoncan/files/2517646/test.zip)

The error show as below:
2018-10-26 12:04:35 [INFO] Connection: Connection opened
2018-10-26 12:04:35 [ERROR] UdsClient: [UnboundLocalError] : local variable 'val' referenced before assignment
Traceback (most recent call last):
  File "C:\project\2018-09-12-GWM-H6\tools\BootloaderTester\code\V1.1\GWMBootloader\ForTest.py", line 16, in <module>
    BlClient.communication_control(3, 1)
  File "C:\Users\nz4510\AppData\Local\Programs\Python\Python36-32\lib\site-packages\udsoncan-1.3-py3.6.egg\udsoncan\client.py", line 86, in decorated
    return func(self, *args, **kwargs)
  File "C:\Users\nz4510\AppData\Local\Programs\Python\Python36-32\lib\site-packages\udsoncan-1.3-py3.6.egg\udsoncan\client.py", line 636, in communication_control
    communication_type = services.CommunicationControl.normalize_communication_type(communication_type)
  File "C:\Users\nz4510\AppData\Local\Programs\Python\Python36-32\lib\site-packages\udsoncan-1.3-py3.6.egg\udsoncan\services\CommunicationControl.py", line 33, in normalize_communication_type
    communication_type = CommunicationType.from_byte(communication_type)
  File "C:\Users\nz4510\AppData\Local\Programs\Python\Python36-32\lib\site-packages\udsoncan-1.3-py3.6.egg\udsoncan\__init__.py", line 852, in from_byte
    val = int(val)
UnboundLocalError: local variable 'val' referenced before assignment

pylessard, 2018-10-26:
Yeah, that's a bug. Will fix soon

kazenoai, 2018-10-26:
Thanks, I think your project is great.

pylessard, 2018-10-28:
Update on this ; The documentation specify that **CommunicationControl** service expects an object of **CommunicationType** where you specified an integer instead. The underlying code tries to handle a integer input, but fails as there was a bug (partial variable renaming). Since this was officially "not supported", there was no unit test covering this.

[image]

Anyway, I improved the code to handle a integer input and added the relevant unit tests. But for the sake of clarity, I think you should use the **CommunicationType** class, the integer input is a mix of 3 information (subnet + 2 flags).

kazenoai, 2018-10-29:
Thanks, I think I need read document more scrutiny.