pylessard/python-udsoncan #10: Do you have an example of SomeLib

milhead2, 2018-11-16:
Not really an issue as much as a question.  I would like to use your tool to help test a UDS server on a unit we're building.  I'm new to UDS but familiar with CAN and J1939.    

I'll pre-apologize and admit that my python is intermediate at best.

In your example code you pull in the MyCar config object from the SomeLib.SomeCar.SomeModel.  Do you have an example (even a small one) of this module..  Currently I'm only trying to query data by identifier on my server..

pylessard, 2018-11-16:
You're right this is probably missing. I'll update the doc soon.
In the mean time, the answer is : just declare a variable in a file, called SomeLib. Then import it and use SomeLib.MyVar. ;)

Since in Python, everything is an object, you can do the same for a function or a classes. They are all treated the same way by the interpreter

Most parameters in the library are integers.

Don't hesitate if you have any other questions.

milhead2, 2018-11-17:
Greetings!  I got the udsoncan to send some requests and think that my server is responding correctly but I'm still getting errors that I think are configuration related.  Specifically I don't believe that I have the encoder setup correctly for decoding a queried VIN.

I'm not sure what to fill in for the 'data_identifier' dictionary.  When I run the script below 

`
import udsoncan
from udsoncan.connections import IsoTPConnection
from udsoncan.client import Client
from udsoncan.exceptions import *
from udsoncan.services import *

ktm_config  = {
	'exception_on_negative_response'	: True,	
	'exception_on_invalid_response'		: True,
	'exception_on_unexpected_response'	: True,
	'security_algo'				: None,
	'security_algo_params'		: None,
	'tolerate_zero_padding' 	: True,
	'ignore_all_zero_dtc' 		: True,
	'dtc_snapshot_did_size' 	: 2,		# Not specified in standard. 2 bytes matches other services format.
	'server_address_format'		: None,		# 8,16,24,32,40
	'server_memorysize_format'	: None,		# 8,16,24,32,40
	'data_identifiers' 			: {0xf190:""},
	'input_output' 				: {}
}

conn = IsoTPConnection('vcan0', rxid=0x123, txid=0x321)
with Client(conn,  request_timeout=2, config=ktm_config) as client:
    client.change_session(DiagnosticSessionControl.Session.extendedDiagnosticSession)  # integer with value of 3
    print("Requesting VIN")
    vin = client.read_data_by_identifier(udsoncan.DataIdentifier.VIN)
    print("Vin = %s" % vin)

`

I get theexception 

`miller@xw8400 [~/src/linux-uds-server] 
$ python3.7 udsClient1.py 
Requesting VIN
[UnexpectedResponseException] : service execution returned a valid response for service ReadDataByIdentifier, but unexpected. Details : Server returned values for data identifier 0x3132 that was not requested and no Codec was defined for it. Parsing must be stopped. 
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/udsoncan-1.4-py3.7.egg/udsoncan/client.py", line 318, in read_data_by_identifier
    services.ReadDataByIdentifier.interpret_response(response, **params)
  File "/usr/local/lib/python3.7/site-packages/udsoncan-1.4-py3.7.egg/udsoncan/services/ReadDataByIdentifier.py", line 102, in interpret_response
    raise ConfigError(key=did, msg='Actual data identifier configuration contains no definition for data identifier 0x%04x' % did)
udsoncan.exceptions.ConfigError: Actual data identifier configuration contains no definition for data identifier 0x3132

During handling of the above exception, another exception occurred:
`

My server has the VIN as "12345678901234567", Here is the dialog as seen from the server side.

`db_init() complete..

PDU Data
  0000 10 03                                            ..
received a packet: 2 bytes
calling process
i15765_tx_app: Called, msg=0xa14b1970, status=0x00000000
PDU Resp
  0000 50 03                                            P.
i15765_tx_app: sent 2 bytes
PDU Data
  0000 22 f1 90                                         "..
received a packet: 3 bytes
calling process
i15765_tx_app: Called, msg=0xa14b1970, status=0x00000000
PDU Resp
  0000 62 f1 90 31 32 33 34 35 36 37 38 39 30 31 32 33  b..1234567890123 -- 10
  0010 34 35 36                                         456
i15765_tx_app: sent 19 bytes
`
so it looks like it was sent correctly..

pylessard, 2018-11-17:
Hi Miller,
Indeed, you need to define a Codec to tell the librarie how to encode/decode your DID since UDS leaves the task of defining this to you. Since your data is ascii string, you'll just need something that decode ascii. Then you can assign your codec to your DID.

I will add an example of that in the documentation soon. Note that your codec can be 3 things

1. A class that inherit DIDCodec  `data_identifiers = {0xf190: MyCodec}`
2. A instance of class that inherit Codec  `data_identifiers = {0xf190: MyCodec(something)}`
3. A string that will be fed to struct.pack() and struct.unpack() `data_identifiers = {0xf190: "<H"}`

In your case, you codec would be defined like this. Note that here, I will make a codec that receive a length as parameter so that you can reuse it if you have ASCII data of different lengtrh than 17 bytes.

    class MyAsciiCodec(DidCodec):
        def __init__(self, string_len):
            self.string_len = string_len
    
        def encode(self, string_ascii):
            return string_ascii.encode('ascii')
    
        def decode(self, string_bin):
            return string_bin.decode('ascii')
    
        def __len__(self):
            return self.string_len

Then you do `data_identifiers = {0xf190: MyAsciiCodec(17)}`

Should do the trick.
By the way, I might be adding an AsciiCodec in the library, so make sure to use a different name if you want to avoid conflict on next update :)

pylessard, 2018-11-17:
Oh Sorry,
I read too fast, the library is complaining that it received a response for a DID that it did not requested 3132 which is obviously the following bytes.  

Can you provide a raw can log? The log you showed seems ok, but is coming from your app, above the TP layer, if there's an issue with the isotp layer, maybe the upper layer get bad data. That would be surprising as Simma Software stack (which you seem to be using) works just fine.

milhead2, 2018-11-17:
Greetings!  Thank you for taking a look at my long issue and on a weekend no less!

As you requested here is a raw can log.  As I understand the isotp encoding it looks ok with the server issuing the long VIN response.

miller@xw8400 [~/src/linux-uds-server] 
$ candump vcan0                                               --Millers Interpretation--
  vcan0  321   [3]  02 10 03                                ...set extended diag
  vcan0  321   [4]  03 22 F1 90                           ...hey server give me F190 (aka VIN)
  vcan0  123   [8]  10 13 62 F1 90 31 32 33       .. here ya go client, 19 bytes, resp, F190, data
  vcan0  321   [3]  30 00 00                                ... client here, please send the rest
  vcan0  123   [8]  21 34 35 36 37 38 39 30       .... more VIN data
  vcan0  123   [7]  22 31 32 33 34 35 36            .... final VIN data

I'll try the rest of your commets and v1.5 over the weekend.

Thank you again for your help.  

Miller

pylessard, 2018-11-17:
You raw can log seems good to me! 
You have been too quick at reading my answer. I edited it 5 minutes afterwards. Please re-read. The issue was in the definition of the codec.

Good luck!

milhead2, 2018-11-17:
Digging deeper, in the candump above,  my CAN server is dropping a byte at the end..  the stack should be sending 20 bytes to the client, 17 bytes of VIN, 2 of DID and 1 response code....      I hand the stack handler 17 bytes, not sure where one is getting dropped but I need to fix this first...  Stay tuned....

milhead2, 2018-11-17:
Woo Hoo!   Only about 75 objects to go...  Hopefully I'll speed up some.
$ python3.7 udsClient1.py 
Requesting VIN
interpret_response: responsedata = b'\xf1\x9012345678901234567', len=19, offset=0
MyAsciiCodec: decoding b'12345678901234567'
interpret_response: responsedata = b'\xf1\x9012345678901234567', len=19, offset=19
VIN=b'12345678901234567'

pylessard, 2018-11-17:
Good work!
I'll close this issue then.

Cheers

closed 2018-11-17 by pylessard

Seppeltronics, 2022-03-23:
Hello Pier-Yves,

your lib seems to be very nice, but the "SomeLib.SomeCar.SomeModel", I do not get it. Could you just post her a proper example for a real existing car? 

What is "SomeLib", are ther decodings for manufacturers such as Renault, Toyota,... so I can directly encode and decode the data for the requests and replies?

Thanks a lot for helping, best Regards, Seppel

pylessard, 2022-03-23:
Hi Seppel,
This package support the Uds protocol as defined by Iso14229. It mainly defines the data container, but not the data content. Each manufacturer define its own DID, routines, dtc, etc.

An analogy with HTTP is possible. Imagine a package that supports http with get, post headers, etc. You're asking how to use the github API. Sire its over HTTP, but its above the protocole itself. The same applies with UDS and Dids, routines, dtc

I do not support specific vehicles with this lib, but you are free to make a repo with some configuration availables for others :)

The documentation is pretty clear about how to configure the client. I can answer most specific questions if you have some.