pylessard/python-udsoncan #11: MemoryLocation memorysize_format

Anthony-Tatowicz, 2018-11-22:
Running udsoncan==1.5 

I found that the memorysize_format param of MemoryLocation doesn't change the format of the data sent on the bus, the only way to get the format to change is by defining it in the config before hand.

Looking into the issue, you may have quick fix though?

pylessard, 2018-11-22:
Interesting. Will look Sunday. This should be covered by unit tests. Can you share a piece of code and a log?

pylessard, 2018-11-22:
I think I found the cause. Copy paste of code that went wrong. 
Will push a fix today. But will release something including unit tests only this weekend.

pylessard, 2018-11-22:
Should be fixed now in branch master. I haven't tested at all, just blindly submitted changes to be quick.
Cheers

Anthony-Tatowicz, 2018-11-23:
Tested, looks good. Ill close out this issue.

closed 2018-11-23 by Anthony-Tatowicz

pylessard, 2018-11-28:
Fixed in v1.6

commit cba00393 (2018-11-22): Fix #11, MemoryLocation was not overriding memorysizze_format