pylessard/python-udsoncan #20: ResponsePending is process but still produces a timeout

XelaRellum, 2019-02-20:
After receiving a couple of ResponsePending respondes a `TimeoutException` is fired. Actually the timeout should be reset.

I have created a fix, but no test case yes (I do have one, but it seems to be too complicated).

pylessard, 2019-02-20:
Hello Alexander,
Thank you for your interest. Actually, this is the intended behaviour as you do not want your software to hangs indefinetely if your ECU keep sending "pending response". A way to exit the wait state is required.

Actually, following #19, I planned into adding a second timeout value. One for a single request and a second global timeout that won't be reset by the 0x78 NRC code.  

I don not have a lot of time this week. This will need to wait a week or two.

XelaRellum, 2019-02-21:
OK, I see. Understand the reasoning behind this. For now I will just increase the overall timeout to the maximum expected pending time and I guess then I am fine...

closed 2019-02-21 by XelaRellum