pylessard/python-udsoncan #22: RuntimeException is not defined in connections.py

J-Wrobel, 2019-03-06:
Hello, I encountered this message while sending request:
`File "/usr/local/lib/python3.5/dist-packages/udsoncan-1.7-py3.5.egg/udsoncan/connections.py", line 68, in wait_frame
    frame = self.specific_wait_frame(timeout=timeout)
NameError: name 'RuntimeException' is not defined`
I am still looking for root cause on my side but I wanted to ask if this exception is correct.
Btw. thanks for very nice lib ;)

pylessard, 2019-03-06:
Hi,
I see the issue. Your connection is not opened and instead of raising a RuntimeError, I raised RuntimeException which does not exists. 

Quick fix is to open the connection first, or use the "with" statement on the client, which opens it.

Will fix soon

J-Wrobel, 2019-03-06:
Hi, yes I used with statement an after closing this statement still wanted to wait for frame by mistake. Thanks for the response.

commit 0906911a (2019-03-06): RuntimeException changed to RuntimeError. Fix #22
closed 2019-03-06 by pylessard