pylessard/python-udsoncan #26: Support for ETAS CAN dongle

strongleap, 2019-05-14:
Hi,
I wanted to know if it is possible to add support for ETAS CAN dongle? 
I have used the PeakCAN and Kvaser interfaces and they work a treat.

I did not see much information online for the ETAS Python interface. I currently hanging on a ETAS ES581.4 dongle without Python support.

Any chance any one has already come across this? Any fix to use it with Python UDSonCAN ?

Many thanks,
Anil

pylessard, 2019-05-14:
Hello, UdsOnCAN doesn't support interfaces, but the underneath layers does. 

Are you able to interract with your hardware with Python?

python-can already has an [issue for this](https://github.com/hardbyte/python-can/issues/189). 
Maybe you can propose a PR to close the issue. If you get Python-CAN to support ETAS itnerfaces,  you'll be able to use them with udsoncan very quickly.  Otherwise, you'll have to make a Connection to interract with it.

Same applies for SocketCAN; if you are willing to improve the module :)

strongleap, 2019-05-15:
Thanks for you quick answer. Yes it makes sense, this is an interface to be sorted out with the python-can. 
I will get the interface and use it with udsoncan.
Definitely will give a try with SocketCAN :-)

closed 2019-05-15 by strongleap

pylessard, 2019-05-15:
Good, don't forget that you don't need python-can or SocketCan. If you can interract with the hardware in Python, you could extend udsoncan as well to use it. But the community will appreciate if you improve python-can.

Cheers

ep081106, 2019-07-10:
> Thanks for you quick answer. Yes it makes sense, this is an interface to be sorted out with the python-can.
> I will get the interface and use it with udsoncan.
> Definitely will give a try with SocketCAN :-)

hello
did you find the solution to use the ETAS ES581.4 ？i want to use the udsoncan by ETAS ES581.43, but i don't know how to connect it. https://github.com/pylessard/python-udsoncan/issues/32 is my problem, could you please give a batter method? thanks!