pylessard/python-udsoncan #29: Cannot use default codec to encode multiple values with client.write_data_by_identifier

pylessard, 2019-05-24:


feversky, 2019-06-26:
change line 41 of "udsoncan\services\WriteDataByIdentifier.py" as below:
`req.data += codec.encode(*value)`

a star unpacks the sequence/collection

closed 2019-08-07 by pylessard

ibutra, 2019-10-15:
Can you push a new Version to pip with this fix (and possibly others?)

pylessard, 2019-10-15:
Hi,
Yes, I will do that soon

ibutra, 2019-10-15:
Thank you very much!
Not only for the new Version but for the whole Package! It helps out a lot!

pylessard, 2019-10-19:
Included in v1.8.1

m31L3r, 2024-11-07:
Hi,
regarding this issue. Why not use in WriteDataByIdentifier.make_request()  function "isinstance(codec, DidCodec)" instead of "codec.__class__==DidCodec"?
That way i could also use this feature for my DidSubclass without having to redefine the encode function.

Regards

pylessard, 2024-11-10:
Here's my thinking.
The DidCodec class, which is the base class have a default behavior to use pack strings as a convenience.  (I should have left that class abstract and make a pack codec, but that is another topic.)   expanding a tuple for a pack string make sense but for a generic codec, I cannot know if the suer wants a tuple or a series of arguments. How would you encode a tuple if the library tries to unpack it? a tuple of tuple?

I think it's better to leave the codec generic and have an explicit implementation

m31L3r, 2024-11-14:
Ah ok I see.
Thanks for your reply!

commit fd0f045e (2019-08-07): Allow using a tuple with write_data_by_identifier and the default codec. Fixes #29