pylessard/python-udsoncan #30: UDS client gets a timeout for when receiving segmented frames while the blocksize within isotp_params is set to zero

VyshakhPV, 2019-06-27:
Hi,

I am using the udsoncan along with python-can-isotp and python-can for communication with my server.  I have set the blocksize for isotp to be zero which should indicate an infinitely large block size. 

But when I try using read_data_by_identifier to get data from the server, the reply for which will be send in more than two segmented frames , an "integer division or modulo by zero" error is also printed out and a TimeoutException is triggered. But I have verified that the server is responding to the request properly.

The log during the same is a provided below

> 2019-06-27 13:41:42 [ERROR] Connection: integer division or modulo by zero
> 2019-06-27 13:41:43 [DEBUG] Connection: No data received: [TimeoutException] - Did not receive frame from user queue in time (timeout=1 sec)
> 2019-06-27 13:41:43 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=1.000 sec)

If I increase the blocksize more than zero, the data is properly received. 

Isn't the above mentioned behaviour a bug ?

pylessard, 2019-06-27:
That looks strange,
Can you share the an example of code or at least a CAN dump?

VyshakhPV, 2019-06-27:
Hello,

Please find an example code which results in the issue.

```
import can
import isotp
import udsoncan.configs
from udsoncan.connections import PythonIsoTpConnection
from udsoncan.client import Client

uds_config = dict(udsoncan.configs.default_client_config)
uds_config['data_identifiers'] = {
    0x0100  : udsoncan.AsciiCodec(11),     # Codec that read ASCII string. We must tell the length of the string
}

udsoncan.setup_logging()    #Uncomment to enable debug logs

isotp_params = {
   'stmin' : 10,                           
   'blocksize' : 0,                     
   'wftmax' : 0,                          
   'll_data_length' : 8,                  
   'tx_padding' : 0,                  
   'rx_flowcontrol_timeout' : 1000,       
   'rx_consecutive_frame_timeout' : 1000, 
   'squash_stmin_requirement' : False
}

bus = can.interface.Bus(bustype='vector', app_name='python-can', channel=1, bitrate=500000) # Link Layer (CAN protocol)
tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x123, rxid=0x456)         # Network layer addressing scheme
stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)                       # Network/Transport layer (IsoTP protocol)
conn = PythonIsoTpConnection(stack)                                                         # interface between Application and Transport layer

with Client(conn, request_timeout=2, config=uds_config) as client:
    response = client.read_data_by_identifier(0x0100)
    print(response.service_data.values[0x0100]) # This is a dict of DID:Value
```

the log for the same is as discussed below

> 2019-06-27 15:32:20 [INFO] Connection: Connection opened
> 2019-06-27 15:32:20 [INFO] UdsClient: ReadDataByIdentifier<0x22> - Reading data identifier : 0x0100 (VehicleManufacturerSpecific)
> 2019-06-27 15:32:20 [DEBUG] Connection: Sending 3 bytes : [b'220100']
> 2019-06-27 15:32:20 [ERROR] Connection: integer division or modulo by zero
> 2019-06-27 15:32:21 [DEBUG] Connection: No data received: [TimeoutException] - Did not receive frame from user queue in time (timeout=1 sec) 
> 2019-06-27 15:32:21 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=1.000 sec)
> 2019-06-27 15:32:21 [INFO] Connection: Connection closed

pylessard, 2019-06-27:
Definetely looks like a bug. I will check that soon.