pylessard/python-udsoncan #31: how can i get the DTC code

ep081106, 2019-07-03:
i used the following code to read the DTC:
with Client(conn, request_timeout=10) as client:
    DTC_list = client.get_dtc_by_status_mask(status_mask)
    print("DTC list:", DTC_list)

but i didn't get the DTC code, i got :
DTC list :<PositiveResponse:[ReadDTCInformation]-61 data bytes at 0x0322ff0>;

could you please tell me how can i get the DTC code?

pylessard, 2019-07-04:
Agree that the documentation is incomplete regarding how the DTC are reported. 
In your case, DTCs are in  ``response.service_data.dtcs`` which is an array of DTC object.

Example 

    response.service_data.dtcs[0].id
    response.service_data.dtcs[0].status
    response.service_data.dtcs[0].Severity

For now, your best source of information will be the unit tests themselves. Each subfunction has a TestSuite associated.  

https://github.com/pylessard/python-udsoncan/blob/7c37ed8b5130f89752d6e33361f4a7849ce9fcf1/test/client/test_read_dtc_information.py#L126-L139

I will try to update the documentation on the short term to make this clearer.

ep081106, 2019-07-04:
Sorry for trouble you again！
Based on your suggestions，My friend and me try our best to find the DTC list, but still cannot get them, maybe our methods are wrong.
we tried to use "print(response.service_data.dtcs)" to get the DTC list, but got the wrong message "name 'resonse' is not defined". 
And we tried copy the code in "python-udsoncan/test/client/test_read_dtc_information.py" Lines 126 to 139, still worng.

could you please give us an example to find it? we have no any method now. thanks!
`
[UDSdemo_0704.zip](https://github.com/pylessard/python-udsoncan/files/3359201/UDSdemo_0704.zip)

`

pylessard, 2019-07-04:
The error is simple. As it says, response doesn't exist. That's because in my example, response is what you get from get_dtc_by_status_mask().  Every client function return a ``Response`` object. In that response you have:
 - ``response.data`` : The raw data of the response
 - ``response.service_data`` : The data interpreted by the right service parser.

In the below image, you don't use the right variable.
[image]

Just do:

    response = client.get_dtc_by_status_mask(status_mask)
    print(response.service_data.dtcs)        # Will print an array of object
    for dtc in response.service_data.dtcs:
        print("DTC : %06X" % dtc.id )         # Print the DTC number

pylessard, 2019-07-05:
Hi,
Did you make it work? Can we close this issue?

ep081106, 2019-07-08:
could you please add "how to read the DTC sanpshot" also？ thanks！

pylessard, 2019-07-22:
Hi, I will leave this issue opened. Documentation needs to be updated. 

Thanks for pointing the issue

jwdsoft, 2020-01-16:
Can you please tell me which hardware could I use in order to use this library ?
also I want to know if this library support trucks also ? since I will test it on a truck ECU
(I did't want to open an issue about these question since they are not issues)

pylessard, 2020-01-17:
Hi @jwdsoft , I think your question deserve a new issue. I've created #44 to help answers.
Honnestly, I personnaly am not aware of what commercial platform uses it. I know about some electric buses and delivery trucks; UDS is trending in the world of EV.

Heavy trucks mostly uses J1939. Hopefully, #44 will gather some useful informations to answer this.

ep081106, 2020-12-29:
hi pylessard, i try to read the DID by using ReadDataByIdentifier, but i get the error: "ConfigError – If didlist parameter or response contains a DID not defined in didconfig";
form the document, i found the example for DidConfig:

**Example of DidConfig**
.. code-block:: python
   didconfig = {
      0x1111 : '<H', # Strings are processed by struct.unpack
      0x2222 : MyCustomDidCodec, # Inherits the udsoncan.DidCodec,
      0x3333 : MyCustomDidCodec(param1='hello') # Instance can also be provided
      0x4444 : dict(key1='val1', key2='val2', codec=MyCustomDidCodec} # If dict is given, a key named "codec" will be searched for
   }

i think the DidConfig shall be added in "udsoncan/__init__.py/class DidCodec", but i still don't know how  to add them;
could you please tell me where shall i add those codes and what things shall i take care (e.g. D030 has 4 bytes with 2、3 or 4 siangls)? thanks!

pylessard, 2020-12-29:
Hi,
This issue is dedicated to DTC. If you have additional problems, creates new issues.
There is no error message that says ``If didlist parameter or response contains a DID not defined in didconfig``

The didconfig must be given to the client.  You should start by looking at the documentation before looking at the code.
There is an example here :   https://udsoncan.readthedocs.io/en/latest/udsoncan/examples.html#reading-a-did-with-readdatabyidentifier

Regards

MasterCodeIT, 2021-07-15:
> The error is simple. As it says, response doesn't exist. That's because in my example, response is what you get from get_dtc_by_status_mask(). Every client function return a `Response` object. In [...]

Still trying to work on this DTC service. So I used the following: 

```
response = client.get_dtc_by_status_mask(status_mask)
        print(response.service_data.dtcs)  # Will print an array of object
        for dtc in response.service_data.dtcs:
            print("DTC : %06X" % dtc.id)  # Print the DTC number
```

any I get this error below:

**response = client.get_dtc_by_status_mask(status_mask)
NameError: name 'status_mask' is not defined**

Kaeptn-G, 2022-09-24:
> Still trying to work on this DTC service. So I used the following:
>
> ```
> response = client.get_dtc_by_status_mask(status_mask)
>         print(response.service_data.dtcs)  # Will print an array of [...]

Is this maybe because you did not define 'status_mask'?
I think you might want to have a look here: https://automotive.wiki/index.php/ISO_14229#DTC_Status_Byte
Example: in case you like to get any DTC, regardless of the status, just define 'status_mask' as 0xff. If you like to see only DTC which have actually been confirmed to be stored non volatile, take 0x8. If you like to only see DTC which are currently failed, take 0x1. 
Is this what you were missing?