pylessard/python-udsoncan #33: Messages using 'suppress_positive_response' are not sent.

GGorany, 2019-07-11:
First, I want you to know that I am not good at English.
so, I ask you a little patience.

I wanted to change the diagSession using the following code.
[image]

But, the message was not sent. I attached logs below.
[image]

As you already know, IsoTP did not work at all.
So, I tried to debug this.

When I set a breakpoint at the location shown in the picture below of 'client.py', a message was sent.
[image]

If I do not set a breakpoint, the message will not be sent.

I use vector hardware in Windows environment.

Am I misusing the API?

Thank you.

pylessard, 2019-07-11:
I see,
As we don't expect a response, the data is sent to the isotp layer and then continue to execute, not waiting for a response. I assume your software exit right after sending the request. Therefore, you may exit before the isotp thread has time to execute.

Quickest fix would be a small delay after the request. But I think there might be some improvement to be done in the stack. Either add a method to explicitly wait for data transmission to be complete or wait inside the client call.

What type of Connection do you use?

GGorany, 2019-07-11:
I solved this problem with a 100ms time delay.
Thank you for your help.

I use PythonIsoTPConnection. ( I hope right answer. )

Your project really helps me a lot.
I would like to say thank you.

pylessard, 2019-07-11:
Ok, you can have a cleaner solution by using the isotp stack "transmitting" method.

    while stack.transmitting():
        Time.sleep(10)

See : https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#usage

pylessard, 2019-10-19:
Decided to close that issue as this problem was not coming from the package itself. Modifying the package to add a feature to detect if the connection is busy would be over-engineering in my opinion.

closed 2019-10-19 by pylessard