pylessard/python-udsoncan #35: Use spaces instead of tabs

danielhrisca, 2019-08-06:
Hello Pier,

first thank you for your work on this package!

I noticed that you use tabs instead of spaces for indentation throughout the codebase. Is it possible to switch to spaces?

pylessard, 2019-08-06:
Hi Daniel,
I am not against it, I just need a reason for it. Why do you need this?

danielhrisca, 2019-08-06:
It's simple really: most users set up their IDE/editors to replace tabs by spaces (most use 4 spaces / tab).

I was having problems running UDS commands over CAN-FD and needed to do some debugging (prints :D) and this really hampered my progress.

christiansandberg, 2019-08-06:
It is generally a good idea to follow some coding guidelines, and PEP-8 is the official so there is not much reason to make up your own. The tabs were an issue for me too.

danielhrisca, 2019-08-06:
@christiansandberg can you have a look at this issue https://github.com/pylessard/python-udsoncan/issues/36

pylessard, 2019-08-06:
That's all I needed
[image]

I didn't really made my own, I simply didn't read PEP-8 and use my editor default setting.

commit 293787d2 (2019-08-07): Fixed indentation with autopep8 --select E101,E11,E121. Fixes #35
closed 2019-08-07 by pylessard