pylessard/python-udsoncan #36: CAN-FD using Kvaser interface

danielhrisca, 2019-08-06:
Not sure if this should be addressed here or in python-can but using Kvaser and CAN-FD the messages are are send as regular CAN not CAN-FD. 

I've created the _bus_ with _fd=True_ but this was not used when sending the final message; the transport layer created regular CAN messages that get passed lower to the interface driver.

pylessard, 2019-08-06:
Have you told your IsoTP layer to set the is_fd parameter to True as well?

See https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#can_fd

CanMessage objects has a  ``is_fd`` property that must be set. Either you set them in your txfn function or you set the ``can_fd`` parameter to true so that the Outputted CanMessage have this property set by default.

See
https://github.com/pylessard/python-can-isotp/blob/03115c07ee754954e4190d5ce4e2241985fef783/isotp/protocol.py#L692-L694

danielhrisca, 2019-08-06:
I don't have my code at hand, but I don't think I have set this parameter. I will see for sure tomorrow

danielhrisca, 2019-08-07:
It works correctly with can-isotp 1.5. Thank you!

closed 2019-08-07 by danielhrisca