pylessard/python-udsoncan #37: Is it possible to send a invalid request

sgnes, 2019-09-06:
Sometimes you want to test the NRC code for UDS service, like for $22, normally the did length is 2, like 03 22 01 01 xx xx xx xx.
if you want to test the NRC 0x13(incorrect message length), you have to send something like 02 22 01 xx xx xx xx xx, and you expect a NRC 0x13.

Is it possible to add this kind of feature?

pylessard, 2019-09-06:
So you would like a way to tamper the payload going out of the client?

sgnes, 2019-09-06:
Thanks for your feedback, in my case,  it is not only request a wrong did(NRC 31),  but also the request length is not correct, wrong format(NRC 13), I have similar test for other services.
I will try with current implementation, and feedback later next week,  thanks.

pylessard, 2019-09-06:
Ok, so the client will (should) always generate valid requests.  

You can directly use the Services object to generate your payload, tamper them then send them.  For example, you can call req = [udsoncan.services.ReadDataByIdentifier.make_request(...)](https://udsoncan.readthedocs.io/en/latest/udsoncan/services.html#readdatabyidentifier-0x22) to get a request, then you can call [req.get_payload()](https://udsoncan.readthedocs.io/en/latest/udsoncan/request_response.html#udsoncan.Request.get_payload) to get a bytes object that you can tamper before sending to the connection.

In other word you kind of redo the Client job.

In that sense, I think it could be a good idea to introduce some sort of middleware to tamper the payload for these type of testing.  

What do you think?

sgnes, 2019-09-08:
One proposal is to add a new interface to send a raw request,  pass into the raw data,  without length information, and return the response to user. 
In this way,  no change for current implementation, and this raw interface can used for testing.

sgnes, 2019-10-29:
Sorry for the delay,  just want to know whether create a new raw request interface is ok for you or not,  if it is ok,  I will create a PR for this,  thanks.

pylessard, 2019-10-30:
How would this interface be different than the Connection object?
I wanted to work on this, but time is lacking on my side.

sgnes, 2019-10-30:
This interface take only one string  parameter, like if you want to change to defaut session, the string will be "1001", and also return the raw data got from server, only handle the NRC.
if this is ok, I will try to create a PR,  because also I don't have sufficient time recently, thanks.

sgnes, 2019-10-31:
Yes, not only just add a new function call,  but add a raw request service class, also a raw request class.
Because I don't suggest to change every services to send a invalid diagnostic requst, better keep current implementation, created some new functionality for invalid requst.
 
The scenario:
For change session service,  we need to test all the positive and negative response, so we want to send something like:
1. 021001        Positive , this one is ok for now.
2. 0110            Negative 
3. 03100101   Negative
4. 021009       Negative, this one is ok for now.

Not sure this "client.payload_modifier(myFunc)" can archive this,  and one more question is how can the user use this one to send a invalid request, for different cases, at least we have two cases,  message lengh is shorter than expected, or longer.

pylessard, 2019-10-31:
I've added the interface as proposed. You do : 

    with self.payload_override(b'\x12\x34\x56\x78'):
        self.client.change_session()  # client will output 12345678 (hex)

Or even

    def my_func(payload):
        return payload + b'\x99'
    
    with self.payload_override(my_func):
        self.client.change_session(55)  # output 105599 instead of 1055

What do you think?

commit 08bb81fe (2019-10-31): Add an interface to override to output payload of the client. Fix #37
commit 23689ab0 (2019-11-01): Updated doc to include new feature that fixes #37
closed 2019-10-31 by pylessard