pylessard/python-udsoncan #40: Timeout issue

sgnes, 2019-11-06:
Please check below issue, let me know if you need more information.
the log:
2019-11-06 09:31:10,654 - UdsClient - ERROR - [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=0.978 sec)
The trace:
[image]
The config:
[image]

sgnes, 2019-11-06:
Another log:

2019-11-06 09:41:07,954 - UdsClient - DEBUG - Sending request to server
2019-11-06 09:41:07,954 - Connection - DEBUG - Sending 4 bytes : [b'14ffffff']
2019-11-06 09:41:07,955 - UdsClient - DEBUG - Waiting for server response
2019-11-06 09:41:08,004 - isotp - DEBUG - Receiving : <1111> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,004 - isotp - DEBUG - Receiving : <1112> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,004 - isotp - DEBUG - Receiving : <1110> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,004 - isotp - DEBUG - Receiving : <1111> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,005 - isotp - DEBUG - Receiving : <1112> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,005 - isotp - DEBUG - Receiving : <1110> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,005 - isotp - DEBUG - Receiving : <1111> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,005 - isotp - DEBUG - Receiving : <1112> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,005 - isotp - DEBUG - Receiving : <1115> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,005 - isotp - DEBUG - Receiving : <1116> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,006 - isotp - DEBUG - Receiving : <1117> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,006 - isotp - DEBUG - Receiving : <1110> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,006 - isotp - DEBUG - Receiving : <1111> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,006 - isotp - DEBUG - Receiving : <1112> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,006 - isotp - DEBUG - Receiving : <1110> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,006 - isotp - DEBUG - Receiving : <1111> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,007 - isotp - DEBUG - Receiving : <1112> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,007 - isotp - DEBUG - Receiving : <1110> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,007 - isotp - DEBUG - Receiving : <1111> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,007 - isotp - DEBUG - Receiving : <1112> (8)	 b'ffffffffffffffff'
**2019-11-06 09:41:08,008 - isotp - DEBUG - Sending : <7E1> (8)	 b'0414ffffff000000'**
2019-11-06 09:41:08,059 - isotp - DEBUG - Receiving : <1110> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,059 - isotp - DEBUG - Receiving : <1111> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,059 - isotp - DEBUG - Receiving : <1112> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,059 - isotp - DEBUG - Receiving : <1110> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,059 - isotp - DEBUG - Receiving : <1111> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,060 - isotp - DEBUG - Receiving : <1112> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,060 - isotp - DEBUG - Receiving : <1113> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,060 - isotp - DEBUG - Receiving : <1114> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,060 - isotp - DEBUG - Receiving : <1115> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,060 - isotp - DEBUG - Receiving : <1116> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,060 - isotp - DEBUG - Receiving : <1117> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,061 - isotp - DEBUG - Receiving : <1118> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,061 - isotp - DEBUG - Receiving : <1119> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,061 - isotp - DEBUG - Receiving : <111A> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,061 - isotp - DEBUG - Receiving : <111B> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,061 - isotp - DEBUG - Receiving : <111C> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,061 - isotp - DEBUG - Receiving : <111D> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,061 - isotp - DEBUG - Receiving : <1110> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,062 - isotp - DEBUG - Receiving : <1111> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,062 - isotp - DEBUG - Receiving : <1112> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,062 - isotp - DEBUG - Receiving : <1110> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,062 - isotp - DEBUG - Receiving : <1111> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,062 - isotp - DEBUG - Receiving : <1112> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,062 - isotp - DEBUG - Receiving : <111E> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,062 - isotp - DEBUG - Receiving : <111F> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,063 - isotp - DEBUG - Receiving : <1120> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,063 - isotp - DEBUG - Receiving : <1121> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,063 - isotp - DEBUG - Receiving : <1122> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,063 - isotp - DEBUG - Receiving : <1123> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,063 - isotp - DEBUG - Receiving : <1124> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,064 - isotp - DEBUG - Receiving : <1125> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,064 - isotp - DEBUG - Receiving : <1126> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,064 - isotp - DEBUG - Receiving : <1127> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,064 - isotp - DEBUG - Receiving : <1128> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,064 - isotp - DEBUG - Receiving : <1129> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,064 - isotp - DEBUG - Receiving : <112A> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,065 - isotp - DEBUG - Receiving : <112B> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,065 - isotp - DEBUG - Receiving : <1110> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,065 - isotp - DEBUG - Receiving : <1111> (8)	 b'ffffffffffffffff'
2019-11-06 09:41:08,065 - isotp - DEBUG - Receiving : <1112> (8)	 b'ffffffffffffffff'
[... cut ...]
2019-11-06 09:41:08,955 - UdsClient - ERROR - [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=0.886 sec)
2019-11-06 09:41:08,956 - UdsClient - ERROR - [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=0.886 sec)
2019-11-06 09:41:08,956 - UdsClient - ERROR - [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=0.886 sec)
2019-11-06 09:41:08,956 - UdsClient - ERROR - [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=0.886 sec)

pylessard, 2019-11-06:
I understand from your config and log that you expect the client to wait an additional 5 sec after NRC 78 and it doesn't, is that it?

Please explain a little bit

sgnes, 2019-11-06:
Yes, you are right, if my understanding is correct, after you receive  NRC 78, it will use the p2 * server, which is 5s.

pylessard, 2019-11-06:
Do you have many client using the same connection? Or different thread using the same client?
There is many timeout error at the end, this should not happen.

Also, can you give me details about your environment?

1.  OS type and version
2. Python version
3. UdsOnCan version and isotp module version

Ideal is that you share your code.

pylessard, 2019-11-06:
Also, in your log, line 71, I see this : 

``now set to P2 (5.000 seconds)*``

The code is 

``  is now set to P2* (%.3f seconds)``

Note the position of the star.

Have you modified the client code?

sgnes, 2019-11-07:
@pylessard 
No, only one client, one connection.

Windows 10 64bit 
Python 3.7 32bit 
Udsoncan 1.9
Other module installed this week,  using pip,  should be the latest version.

pylessard, 2019-11-07:
Ok.
Are you able to explain the 5 extras TimeoutException at the end of the log and also the line 71 that does not match the code in the repo?

sgnes, 2019-11-07:
you can close this one, seems the issue is when I init the Client, I passed request_timeout with 1, which overwrite the config in the client config, thanks.

closed 2019-11-07 by pylessard