pylessard/python-udsoncan #41: problem with multiple PythonIsoTpConnection open

IDC-Dragon, 2019-11-19:
I'm seeing a problem when having multiple PythonIsoTpConnection objects open() at the same time. Sending a read_data_by_identifier() hangs most of the time, but not always, statistically more often the more connections are open. They seem to interfere / block each other.
Symptom is, the ECU sends back the FirstFrame of the answer, but the host is not replying with a FlowControlFrame. Communication ceases at this point, in consequence no ConsecutiveFrames get sent by the ECU, the request finally times out.
I've observed this with v1.8 and v1.9, on different hosts (ARM and Intel) and different Python versions (3.55 and 3.74)
When using the kernel native IsoTPSocketConnection instead, it works fine. However, I can't use this since it requires Python3.7.

pylessard, 2019-11-19:
Interesting.
Can you share sone code? Or at least your configuration

IDC-Dragon, 2019-11-19:
I was digging a bit deeper meanwhile.

Having multiple PythonIsoTpConnection open means each has a running rxthread_task(). They seem to compete for the received CAN payload. What I see is that a call to self.isotp_layer.process() discards the packet because of address mismatch. So, a thread for another ECU has fetched the FirstFrame.

Am I using this conceptually wrong? Is it meant to allow multiple active connections? In that case, the receiver socket should have a filter, or one central receiver thread has to dispatch the packets to the instances.

pylessard, 2019-11-19:
That's a very good question. I think each of your bus object should filter for the correct ID so that they do not consume the CAN message for other connections (I am assuming that this is the behaviour python-can when setting a filter). 

Having a central dispatcher might be the most straight-forward option. You can use the QueueConnection for that, but I think correctly filtering is cleaner.

Maybe it would be relevant that a filter is automatically set based on the IsoTP stack parameters. 
I will investigate later. 

I'd be curious to hear what @christiansandberg think of this.

In any case, sharing your configuration may help.

IDC-Dragon, 2019-11-19:
I have now changed my code to create an individual bus object per ECU, with a filter set to the Rx address. This works for me.
A central dispatcher might be lighter to the CPU, but is more difficult to code. IIRC the tread has short timeouts, wakes up often. With many of those, it's a lot of polling.

pylessard, 2019-11-19:
Thanks for the feedback.
Do you think udsoncan or can-isotp should be altered in any way to better handle that use case?

IDC-Dragon, 2019-11-20:
Dunno, my case might be exotic. Maybe the documentation tell the caveat. I was especially puzzled because IsoTPSocketConnection worked, PythonIsoTpConnection didn't.
Their setup syntax is different, however. IsoTPSocketConnection gets a bus name and can do all the socket setup by itself, vs. PythonIsoTpConnection operates somewhat lower. You can't set filters or other sockopts when the socket is already bound.
Nevermind, IMHO you may close this. Thanks a lot for the fine piece of Python!

pylessard, 2019-11-20:
PythonIsoTpConnection is pretty much a shortcut for not having to setup the socket. You can always use the SocketConnection if you want to do some fancy sockopt setting. I don't think your setup is exotic, I think it would be common to have many UDS channel on the same CAN bus.

And you are correct, the handling of message is different in the kernel module v.s. what python-can does. When using the PythonIsoTpConnection, there an instance of socket per connections and the kernel-module does the dispatching to give a copy of the CAN messagse to each socket unless there's a filter. You'd have to code a dispatching mechanism yourself to get something similar with python-can.

I'll add a warning in the documentation regarding this potential issue. I might also add an function call to get the correct mask from the address object. something like ``my_address.get_can_filter_mask()``

Will close this issue
Thanks for the input.

closed 2019-11-20 by pylessard