pylessard/python-udsoncan #46: Usage on windows with any hardware

jwdsoft, 2020-01-27:
I want to use this library to talk to an ECU.I saw on the examples that its using can0 (that means it is used on linux where you will setup a can device as can0) 
But I want to use it in windows.what should i use instead of can0 ? And which hardware to use ? Can i use an arduino for example with a particular sketch or use a J2534 device ...
I don't know what to do i can't use it on linux since the software that talk to the ECU is windows. And i can't switch back and forward between linux and windows
Any idea on how to get up and running on window? s

pylessard, 2020-01-27:
Hi,
UDS is only one layer (application layer) in the communication stack. It is normally used over a transport layer, usually IsoTP (ISO-15765). This project only handles the UDS part ([there is a another repo to that handle IsoTP](https://github.com/pylessard/python-can-isotp)). It is not dependent on the operating system, you have the possibility to build the underlying layers yourself.

The way this project handle underlying layers is through an object of type ``Connection``. See the [documentation ](https://udsoncan.readthedocs.io/en/latest/udsoncan/connection.html)on the subject. You can always write your own connection to handle your specific case or use one of the generic connections that write to a FIFO or a Socket and write the other side of the pipe.

Assuming you use the IsoTP protocol as a transport layer and you use my other project, you can use the ``PythonCanIsoTPConnection`` object. This object will bridge the UDS layer with the IsoTP layer. Your problem will be pushed down a layer below and yuou will need to find a way to bridge the IsoTP layer with your hadware.  There, you'll have 2 options

 1. Write 2 functions, one for reception (``rxfn``) and one for transmission (``txfn``) and give these functions to the IsoTP layer. You make the bridge between the transport layer and the link layer
 2. Make usage of [python-can](https://python-can.readthedocs.io/en/master/) that is a cross-platform CAN library that works under Windows with the most popular dongle out there . The bridge between IsoTP layer and python-can is already written, check for the ``[CanStack`` object](https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#isotp.CanStack). If your device is supported by python-can, it should be a no brainer to setup.

Also, note that any implementation of the underlying layers (transport/link layer) executed in the user space won't be able to respect the timings dictated by ISO-15765. Only the Linux kernel module and some other vendor specific libraries can guarantee their respect.

Finally, take the time to read the documentation. Most of what I wrote is already written in there.

Regards

pylessard, 2020-01-30:
Can this issue be closed?

jwdsoft, 2020-01-30:
Thanks you my friend.i should've fully read the docs before creating this issue

closed 2020-01-30 by jwdsoft