pylessard/python-udsoncan #47: Support RequestFileTransfer (0x38) service

wanam, 2020-03-20:
Hello @pylessard ,

Just wanted to know if you have any plans to support `RequestFileTransfer (0x38)` service, which is defined on `ISO_FDIS_14229-1`.

pylessard, 2020-03-20:
Hi @wanam ,
Yes, I learned recently that this was added to the latest standard. Do you need it?

wanam, 2020-03-20:
Yes, indeed we need it.

pylessard, 2020-03-20:
Alright, will get on it soon

wanam, 2020-03-20:
Nice! thank you.
Let me know if you need any information.

wanam, 2020-03-20:
Hello @pylessard,
I know it's your free time and I should not ask for ETA :) , but do you have any idea about the amount effort needed for such implementation?

Thanks again and have a good day.

pylessard, 2020-03-20:
This won't be too long. I'll get the new standard early next week. Should be done by the weekend after that, so around March 28th. Sounds good?

wanam, 2020-03-20:
Yes it should be fine, thanks a lot for your prompt response.

pylessard, 2020-03-21:
Out of curiosity, are you using the 2013 or 2020 version of the standard? I just learned that a new version was published few weeks ago

wanam, 2020-03-21:
I'm using the old one `ISO/FDIS 14229-1:2012(E)`.

pylessard, 2020-03-24:
For your information, I've started working on this. This seems straightforward

wanam, 2020-03-25:
I saw that you pushed some tests for this service on a RequestFileTransfer branch.

FYI a friend has pushed draft here: https://github.com/wanam/python-udsoncan/commit/fa1457be53a0f0bddbe17c32a595c4a54e4ffc3d

pylessard, 2020-03-25:
Thanks,
I indeed pushed some test. The draft seems well written, but will of course conflict with the actual test as we do not have the same naming. I will refer to it to make sure I don't forget something.

pylessard, 2020-03-28:
Completed. Now available in v1.11
Documentation is up to date.  Let me know what you think

wanam, 2020-03-30:
Well done! thank you.
Will test and get to you back.

pylessard, 2020-04-01:
Can this be closed?

wanam, 2020-04-01:
Sorry for late reply, yes you can close it.
Thank you again.

closed 2020-04-01 by pylessard