pylessard/python-udsoncan #49: Passing Security Access level to security_algo

martinjthompson, 2020-03-27:
In order to handle the case when a project has different algos for different levels, I need to be able to pass which access level is being unlocked to my security_algo. 

It seems that I cannot currently do that, is that correct? Or have I missed how to do this through the `params` parameter?

pylessard, 2020-03-27:
You are correct, I didn't forsee that use case. This should be added.
In the mean time, you can switch algo before doing the call as a workaround.

    client.config['secutirty_algo'] = algo_dict[security_level]
    client.unlock_security_access(security_level)

pylessard, 2020-03-29:
Fixed in V1.12.
Documentation is updated.  security_algorithm will now be given the security level.
For backward compatibility, named argument are now used and only parameters present in the algorithm signature will be given.

Let me know if you encounter any issue with this.

closed 2020-03-29 by pylessard

martinjthompson, 2020-03-30:
Thanks - that's a speedy response :)

commit 8b65f136 (2020-03-29): Now pass the security level to the security algorithm in client.unlock_security_access. Fixes #49