pylessard/python-udsoncan #50: filePathAndNameLength shall be encoded on 2 bytes for 0x38 service

wanam, 2020-04-02:
Hello @pylessard ,

According to the ISO, in service 0x38, the parameter “filePathAndNameLength” should be encoded on 2 bytes in the Request Message Definition.

[image]

This is not the case in your implementation, on RequestFileTransfer.py file, [here](https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/services/RequestFileTransfer.py#L82-L84).

``` python
        path_ascii = path.encode('ascii')
        if len(path_ascii) > 0xFF:
            raise ValueError('Path length must be smaller or equal than 255 bytes when encoded in ASCII')
```

Max path length parameter should be 0xFFFF (65535 bytes).

As a consequence [here](https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/services/RequestFileTransfer.py#L115), since our path is encoded in less than 0xFF, we continue to build a corrupted request:
``` python
        data += len(path_ascii).to_bytes(1, 'big')
        data += path_ascii](url)
```
So the first byte of the path will be stripped on the ECU side as the second byte of path length.

I hope the issue is clear, let me know if you need any further information.

pylessard, 2020-04-02:
Will verify and fix today

pylessard, 2020-04-02:
All done, 
Release is published : v1.12.1

Thank you for your feedback

commit 1294641f (2020-04-02): Fixed #50 - Correctly encode filename length in RequestFileTransfer
closed 2020-04-02 by pylessard