pylessard/python-udsoncan #51: SecurityAccess does not handle already-unlocked server

awilke, 2020-04-17:
If a UDS server is already unlocked for a given security level, then it responds to subsequent requestSeed requests for that security level with an all-zero seed value (see ISO 14229-1:2020 Section 10.4.1).

However, `udsoncan` doesn't appear to handle this case and attempts to send a key value based on this zero value. This causes the server to respond with NRC requestSequenceError. Instead, if the client receives `securitySeed == 0`, then it should early return with success without sending a sendKey request.

pylessard, 2020-04-18:
Thank you very much for the clear and detailed bug report.
I missed that fact when I read the standard. 

This issue is already fixed and the fix is published in v.1.12.2

Regards

closed 2020-04-17 by pylessard

awilke, 2020-04-18:
Wow, super fast response. Thanks for the fix!

commit 138cfcdd (2020-04-17): Fixes #51. Correctly handle unlocking security access level that is already unlocked (seed=0)