pylessard/python-udsoncan #52: TimeoutException from client.read_data_by_id() and client.clear_dtc()

calibramicke, 2020-05-05:
Hi!

I am having a problem where I get TimeoutExceptions when I'm trying to test my small client program against an UDS-server. Both read_data_by_identifier() and clear_dtc() gets the TimeoutException. I know that the UDS-server is working since another program can send and receive UDS messages, so I must be doing something wrong. 

The log gave me the following output from three UDS requests I sent, but every other UDS request gave me similar output, sometimes with "[ERROR] Connection: The receive queue was read too late" and sometimes without it.

2020-05-05 15:01:14 [INFO] Connection: Connection opened
2020-05-05 15:01:14 [INFO] UdsClient: ReadDataByIdentifier<0x22> - Reading data identifier : 0x2b12 (VehicleManufacturerSpecific)
2020-05-05 15:01:14 [DEBUG] Connection: Sending 3 bytes : [b'222b12']
2020-05-05 15:01:15 [DEBUG] Connection: No data received: [TimeoutException] - Did not receive frame IsoTP Transport layer in time (timeout=1 sec)
2020-05-05 15:01:15 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=1.000 sec)
Did not receive response in time. P2 timeout time has expired (timeout=1.000 sec)
2020-05-05 15:01:15 [INFO] Connection: Connection closed

2020-05-05 15:01:32 [INFO] Connection: Connection opened
2020-05-05 15:01:32 [INFO] UdsClient: ClearDiagnosticInformation<0x14> - Clearing all DTCs (group mask : 0xFFFFFF)
2020-05-05 15:01:32 [DEBUG] Connection: Sending 4 bytes : [b'14ffffff']
2020-05-05 15:01:33 [DEBUG] Connection: No data received: [TimeoutException] - Did not receive frame IsoTP Transport layer in time (timeout=1 sec)
2020-05-05 15:01:33 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=1.000 sec)
2020-05-05 15:01:33 [INFO] Connection: Connection closed

2020-05-05 15:02:16 [INFO] Connection: Connection opened
2020-05-05 15:02:16 [INFO] UdsClient: ClearDiagnosticInformation<0x14> - Clearing all DTCs (group mask : 0xFFFFFF)
2020-05-05 15:02:16 [DEBUG] Connection: Sending 4 bytes : [b'14ffffff']
2020-05-05 15:02:17 [DEBUG] Connection: No data received: [TimeoutException] - Did not receive frame IsoTP Transport layer in time (timeout=1 sec)
2020-05-05 15:02:17 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=1.000 sec)
2020-05-05 15:02:18 [ERROR] Connection: The receive queue was read too late
2020-05-05 15:02:18 [INFO] Connection: Connection closed

I was using the PCAN-USB FD interface with the following configuration:

in backend.py
```
 bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=500000)
 config = dict(configs.client_config)

 isotp_params = {
   'stmin' : 32,                      
   'blocksize' : 8,                       
   'wftmax' : 0,                          
   'll_data_length' : 8,                 
   'tx_padding' : 0,                     
   'rx_flowcontrol_timeout' : 10000,       
   'rx_consecutive_frame_timeout' : 10000, 
   'squash_stmin_requirement' : False     

  addr = isotp.Address(isotp.AddressingMode.Normal_11bits, rxid=id, txid=interface_ID) 
  stack = isotp.CanStack(bus=bus, address=addr, params=isotp_params) 
  conn = PythonIsoTpConnection(stack) 

  with Client(conn, config=config) as client: 
        try:
            response = uds_requests.chosen_service(client, input_dict)
            if response != None:
                msg_queue.put(response)

        except Exception as e:
            print(e)

```

In uds_requests.py
```
def chosen_service(client, dict):
     sid = dict.get("SID") #saved as string in dict
    
     if sid == "22":

        did = int(dict.get("DID), 16) #saved as string in dict
        try:
            response = client.read_data_by_identifier(did)
        except Exception as e:
            print(str(e))
        return response

    elif sid == "14":

         data = int(dict.get("DATA"), 16)
          try:
                response = client.clear_dtc(data)
           except Exception as e:
                print(str(e))
                response = None
          return response

```

In configs.py:
```
class MyCustomCodecThatShiftBy4(udsoncan.DidCodec):
   def encode(self, val):
      val = (val << 4) & 0xFFFFFFFF # Do some stuff
      return struct.pack('<L', val) # Little endian, 32 bit value

   def decode(self, payload):
      val = struct.unpack('<L', payload)[0]  # decode the 32 bits value
      return val >> 4                        # Do some stuff (reversed)

   def __len__(self):
      return 4    # encoded paylaod is 4 byte long.

client_config  = {
        'exception_on_negative_response'	: False,
        'exception_on_invalid_response'		: False,
        'exception_on_unexpected_response'	: False,
        'security_algo'				: None,
        'security_algo_params'		: None,
        'tolerate_zero_padding' 	: True,
        'ignore_all_zero_dtc' 		: True,
        'dtc_snapshot_did_size' 	: 2,		# Not specified in standard. 2 bytes matches other services format.
        'server_address_format'		: None,		# 8,16,24,32,40
        'server_memorysize_format'	: None,		# 8,16,24,32,40
        'data_identifiers' 			: {
                                        0x2b12 : MyCustomCodecThatShiftBy4(),    
                                        },
        'input_output' 				: {},
        'request_timeout'			: 10,
        'p2_timeout'				: 2,
        'p2_star_timeout'			: 5,
}

```
Also, I am having trouble to grasp what the DidCodec is supposed to do, I have read the documents and the examples, but I don't get what it is supposed to do. MyCustomCodecThatShiftBy4() does a logical shift left 4 times in encode and then a logical and with 0xFFFFFFFF to keep all 1's. And the decode shifts the response 4 positions to the right.  Whats the point of doing this? What am I missing?

pylessard, 2020-05-05:
Hi,
It looks like no data is being received. Can you share a CAN dump as well as the ID you used for the ISOTP address. 

Also the DIDCodec converts a payload of bytes into a Python object/value. Since they are Manufacturer Specific, you have the freedom of defining your own codecs to interpret customs DIDs the way you want. For example, you could convert a date encoded in ASCII to a Datetime object.  The ``MyCustomCodecThatShiftBy4`` is nothing more than an example. It is not to be used for your application, write your own.

calibramicke, 2020-05-05:
Hi!

Unfortunately, I have limited access to the car that i tested my program on and I didn't take any candumps. But I think i have an idea of where i screwed up. When you mentioned the ID for the ISOTP address I looked at the documentation again and I think I mixed txid and rxid up. "txid - The CAN ID for transmission" was read as the transmitters ID i.e my interfaces id, and "rxid - The CAN ID for reception" was read as the receivers ID i.e the UDS-server.

To answer your question of the IDs  for the ISOTP address, rxid=id was 0x740 (supposed to be the UDS-server address), and txid=interface_ID was 0x2de (the pcan interface id)

Regarding the DIDCodec, I might be slow on this, but am I understanding it correctly that the encode is supposed to convert the data to bytes while decode converts the data from bytes to a object/value? 

Thanks for your answers!

pylessard, 2020-05-05:
Good!
For the codec, you are correct.   It simply is an abstraction layer that you can control. Like this, you don'T have to modify your program if you want to read something from 2 different ECU. For example, you could read VIN from 2 cars that encode their VIN in different fashions by simply loading 2 different configurations into your client.

CAN payload (bytes) -->  codec.decode() --> python object
Python object -->  codec.encode() --> CAN payload (bytes)

calibramicke, 2020-05-05:
Thanks for the fast and pedagogic answer :) 

I'll try the program later this week so I can see if the problem is solved

calibramicke, 2020-05-06:
It seems to be working pretty good now. Thanks for the help!

closed 2020-05-06 by calibramicke