pylessard/python-udsoncan #53: No exception is raised when UDS server respond with no Session Params on Diagnostic Control Session Request

Rohan621, 2020-05-21:
The interpret_response method of the Diagnostic Session service does not raise an InvalidResponseException if the UDS Server respond back to with no session control params. 
https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/services/DiagnosticSessionControl.py#L54

As per ISO-14229(1) Section 10.2, there is no mention of the sessionParameterRecord being optional.

pylessard, 2020-05-21:
Thanks,
It was optional in the 2006 version, which is the standard I initially used to write to library. I will update accordingly. I guess the best approach would be to add a parameter telling what version of the standard needs to be supported.

And what version are you using, in my document, the section is 9.2, not 10.2

Rohan621, 2020-05-22:
Hi,

I am using the 2020 version.

pylessard, 2020-06-19:
I reopened, this need to be fixed. Just didn't have time to do it

Rohan621, 2020-06-24:
Oops. Sorry about that. Thanks for reopening the isse 😄

commit 39e95f0f (2020-07-19): Added support of different standard version.  Enforce reception of timing requirement in DiagnosticSessionControl. Fixes #53
closed 2020-07-19 by pylessard