pylessard/python-udsoncan #55: Send uds command byte 0 padded with 00

mmclwd, 2020-06-12:
While sending UDS service with udsoncan library, is CAN tools is vector it worked fine, but if CAN tools is PCAN, first bytes will padded with 00, see below message information.
Time        Diff Time   Chn     ID    Name     Event Type   Dir    DLC   Data length   Data                          
 [-] 16.733302   0.000000    CAN 1   700   reqUDS   SF           Rx     0     8             [00] [02 3E 00 55 55 55 55]    
UDS protocol is based on CAN not CANFD.
So how to fix this issue? 
Many thanks!

pylessard, 2020-06-12:
Can you share your IsoTP configuration?

mmclwd, 2020-06-15:
Thank @pylessard, I had already fixed this issue by update python-can library, maybe because the python-can lib I used is too old.

closed 2020-06-16 by pylessard