pylessard/python-udsoncan #56: Cannot pass Single Value to InputOutputControlByIdentifier

Rohan621, 2020-06-17:
Unable to pass a single integer value to InputOutputControlByIdentifier service for overwriting a field represented by a Data Identifier. The service only accepts the following inputs:

- A list for positional arguments
- A dict for named arguments
- An instance of :ref:`IOValues<IOValues>` for mixed arguments

but not a single argument such as
```
udsoncan.services.InputOutputControlByIdentifier.make_request(
            did=0xFFF,
            control_param=1,
            values=10,
            ioconfig={0xFFFF: ChildDidCodec()},
        )
```

Getting the following error from this block of code: https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/services/InputOutputControlByIdentifier.py#L76
```
        if values is not None:
            if isinstance(values, list):
                values = IOValues(*values)
            if isinstance(values, dict):
                values = IOValues(**values)
    
            if not isinstance(values, IOValues):
>               raise ValueError("values must be an instance of IOValues")
E               ValueError: values must be an instance of IOValues

```

pylessard, 2020-06-18:
So the code behave like the documentations says it should. I am not sure there is an issue here.
Just add brackets around your value, like this.

    udsoncan.services.InputOutputControlByIdentifier.make_request(
            did=0xFFF,
            control_param=1,
            values=[10],
            ioconfig={0xFFFF: ChildDidCodec()},
        )

Rohan621, 2020-06-19:
Sounds good. Not something super critical. I can close this ticket. Thank you!

closed 2020-06-19 by Rohan621