pylessard/python-udsoncan #57: InputOutputControlByIdentifier does not handle udsoncan.DidCodec.ReadAllRemainingData

Rohan621, 2020-06-17:
The InputOutputControlByIdentifier does not handle `udsoncan.DidCodec.ReadAllRemaining` exception which is raised when the length of a Codec is requested.
https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/services/InputOutputControlByIdentifier.py#L186

This should be handled the same way as in the ReadDataByIdentifier: https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/services/ReadDataByIdentifier.py#L65

pylessard, 2020-06-18:
Correct, I forgot to propagate this feature to InputOutputControlByIdentifier, this could be implemented.  I will look at this soon.

Consider that raising ReadAllRemaining is not a behaviour defined by ISO-14229. Be carefull when using this, what you are trying to achieve may not be complient with the standard.

Rohan621, 2020-06-19:
Thanks! And agreed, the `ReadAllRemainingData` is not a behavior defined in ISO-14229 but is a part of your package. This exception does provide us to bypass calculating the length when dealing with variable length payloads being transmitted and received between the UDS Client and Server.

Looking forward to this change! 😄

closed 2020-07-19 by pylessard

pylessard, 2020-07-19:
Handled in release v1.13.1.  Sorry for the delay

commit d75d8106 (2020-07-19): Support Codec.ReadAllData in IoControl service.  Fixes #57.