pylessard/python-udsoncan #58: No exception raised by WriteDataByIdentifier for Incorrectly echoed Data Identifier

Rohan621, 2020-06-24:
The `interpret_response` method in the WriteDataByIdentifier service does not check if the Data Identifier Value echoed back by the UDS Server is correct or not. It should trigger an InvalidResponseException as done in ReadDataByIdentifier.

https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/services/WriteDataByIdentifier.py#L62

pylessard, 2020-06-24:
Hi, 
This is handled by the client. The service object is not aware of the communication with the server, it just encode and decode the information. It raises a UnexpectedResponseException when this happens.

https://github.com/pylessard/python-udsoncan/blob/3012c716299730c23f58d7e545d5bb22f301d1c7/udsoncan/client.py#L430-L431

Rohan621, 2020-06-24:
Sounds good. Thanks for clarifying!

closed 2020-06-24 by Rohan621

pylessard, 2020-07-19:
Handled in release v1.13.1.  Sorry for the delay