pylessard/python-udsoncan #60: During handling of the above exception, another exception occurred:

WSXXB-main, 2020-07-08:
**this is part of the code that I writed.**

class UdsApp(object):
    def __init__(self):
        self.__bitrates = {
            '500K': 500000,
            '250K': 250000,
            '125K': 125000,
        }

    def usd_app_deal(self,channel,bitrate,isotp_params,PhyID,ResponID):
        # 自定义logging.conf配置文件
        udslog_config_file = path.join(path.dirname(path.abspath(__file__)), 'logging.conf')
        # 开始起动udslogging
        udsoncan.setup_logging(config_file = udslog_config_file)

        thread = Thread(target=self.uds_app, args=(channel,bitrate,isotp_params,PhyID,ResponID))
        thread.start()

    def uds_app(self,channel,bitrate,isotp_params,PhyID,ResponID):
        # Link Layer (CAN protocol)
        bus = KvaserBus(channel=int(channel),bitrate=self.__bitrates[bitrate])
        # Network layer addr

essing scheme
        tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=eval(PhyID), rxid=eval(ResponID))
        # Network/Transport layer (IsoTP protocol)
        stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)
        # interface between Application and Transport layer
        conn = PythonIsoTpConnection(stack)
        # Application layer (UDS protocol)

        with Client(conn, request_timeout=1) as client:
            try:
                client.change_session(DiagnosticSessionControl.Session.defaultSession)
                # client.read_data_by_identifier_first(0xF190)
                # res_f193 = client.read_data_by_identifier(0xF193)
                # print(res_f193.service_data.values[0xF190])  # This is a dict of DID:Value
                # res_f195 = client.read_data_by_identifier(0xF195)
                # print(res_f195.service_data.values[0xF190])  # This is a dict of DID:Value
                # 
                # client.change_session(DiagnosticSessionControl.Session.extendedDiagnosticSession)
                # client.control_dtc_setting(ControlDTCSetting.SettingType.off)
                # client.communication_control(0x01,CommunicationControl.ControlType.disableRxAndTx)
                # 
                # client.change_session(DiagnosticSessionControl.Session.programmingSession)
                # client.request_seed(0x11)
            except NegativeResponseException as e:
                print('Server refused our request for service %s with code "%s" (0x%02x)' %
                      (e.response.service.get_name(), e.response.code_name, e.response.code))
            # except InvalidResponseException or UnexpectedResponseException as e:
            #     print('Server sent an invalid payload : %s' % e.response.original_payload)
            # # except UnexpectedResponseException as e:
            # #     print('Server sent an invalid payload : %s' % e.response.original_payload)
            finally:
                print('其他错误')

**In fact, the can bus donot  response [client.change_session],  But the program reported the following error:**

2020-07-08 14:49:30 [INFO] Connection: Connection opened
2020-07-08 14:58:58 [INFO] UdsClient: DiagnosticSessionControl<0x10> - Switching session to defaultSession (0x01)
2020-07-08 14:58:58 [DEBUG] Connection: Sending 2 bytes : [b'1001']
2020-07-08 14:58:59 [DEBUG] Connection: No data received: [TimeoutException] - Did not receive frame IsoTP Transport layer in time (timeout=0.9980063438415527 sec) 
2020-07-08 14:58:59 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=0.998 sec)
其他错误
2020-07-08 15:02:10 [INFO] Connection: Connection closed
Traceback (most recent call last):
  File "C:\Users\Gary\AppData\Local\Programs\Python\Python38\lib\site-packages\udsoncan-1.12.2-py3.8.egg\udsoncan\client.py", line 1631, in send_request
    payload = self.conn.wait_frame(timeout=timeout_value, exception=True)	
  File "C:\Users\Gary\AppData\Local\Programs\Python\Python38\lib\site-packages\udsoncan-1.12.2-py3.8.egg\udsoncan\connections.py", line 68, in wait_frame
    frame = self.specific_wait_frame(timeout=timeout)
  File "C:\Users\Gary\AppData\Local\Programs\Python\Python38\lib\site-packages\udsoncan-1.12.2-py3.8.egg\udsoncan\connections.py", line 490, in specific_wait_frame
    raise TimeoutException("Did not receive frame IsoTP Transport layer in time (timeout=%s sec)" % timeout)
udsoncan.exceptions.TimeoutException: Did not receive frame IsoTP Transport layer in time (timeout=0.9980063438415527 sec)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Gary\AppData\Local\Programs\Python\Python38\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "D:\01 工作\13 软件开发\04 Python\02 Bootload tool\02 py\V1_2\APP1\UdsApp.py", line 44, in uds_app
    client.change_session(DiagnosticSessionControl.Session.defaultSession)
  File "C:\Users\Gary\AppData\Local\Programs\Python\Python38\lib\site-packages\udsoncan-1.12.2-py3.8.egg\udsoncan\client.py", line 123, in decorated
    return func(self, *args, **kwargs)
  File "C:\Users\Gary\AppData\Local\Programs\Python\Python38\lib\site-packages\udsoncan-1.12.2-py3.8.egg\udsoncan\client.py", line 181, in change_session
    response = self.send_request(req)
  File "C:\Users\Gary\AppData\Local\Programs\Python\Python38\lib\site-packages\udsoncan-1.12.2-py3.8.egg\udsoncan\client.py", line 1639, in send_request
    raise TimeoutException('Did not receive response in time. %s time has expired (timeout=%.3f sec)' % (timeout_name_to_report, timeout_value))
udsoncan.exceptions.TimeoutException: Did not receive response in time. Global request timeout time has expired (timeout=0.998 sec)

**my question:**
1、why isoIsoTP and cilent all reported timeout errors?
2、 why prompt error "During handling of the above exception, another exception occurred:"?

pylessard, 2020-07-08:
Hi.
1. Because you did not receive a response from the server.  Check the CAN log and make sure that you receive data within proper timing with CAN ID that matches your ISOTP stack.

2. Because the client code raise a TimeoutException when the underlying Connection object raises one.  It's just how the code is written.  See https://github.com/pylessard/python-udsoncan/blob/c1094bdd9e9ba6ff085ec63fa1be8ff86f3a4388/udsoncan/client.py#L1632-L1639

I can and will rearrange the code to get a single error message to avoid confusion.

WSXXB-main, 2020-07-09:
Thank you for your answer.

     I can understand ISOTP and Client All Raise TimeoutException because the bus really does not respond to the message.
    As for the second error：the software Pycharm reports an error "During handling of the above exception, another exception"  .    Can be understood as when  ISOTP & Client all Raise TimeoutException, this second error I encountered is normal？  If the answer is YES， Please close the problem。

pylessard, 2020-07-09:
Yes, the second error. You can see in the snippet of code I previously posted that the second exception is raised on purpose.

Will close this issue

closed 2020-07-09 by pylessard