pylessard/python-udsoncan #63: Block Sequence Number for Transfer Data service needs to start with 1 rather than 0

Rohan621, 2020-09-08:
As per ISO-14229(1)-2020 Section 15.4.3.2 Table 451, the Block Sequence Counter value needs start with 1. In the source code https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/services/TransferData.py#L36, the minimum valid value is set to 0. This needs to be updated

pylessard, 2020-09-08:
Hi @Rohan621, 
Can you share the cotnent of that table. ISO-14229:2013 - Section 14.4.2.1 (Table 403) Specifies that the ``blockSequenceNumber`` goes grom 0x00 to 0xFF.  I don't have access to the 2020 version (yet)

Rohan621, 2020-09-11:
Hi @pylessard, I re-read the ISO-14229:2020 again and as per the doc when you start sending data using TransferData the block sequence should start with 1 but that gets rolled over to 0 once the counter passes 255. So the sequence for TransferData looks like this: 1,2,3,4.....,255,0,1,2,3,4....255,0,1,2,3....

Based on the above comment, your source code is working as expected. False alarm on my part. Apologies.

I will close this issue 😄

closed 2020-09-11 by Rohan621