pylessard/python-udsoncan #64: can't get the response.service_data.values

CSG2019, 2020-09-14:
Hello, friends,
Recently I just found this UDS library and it's very useful for me.
But i am not yet familiar with it.
I tried to follow the example in the document to read the$F190.
But i can't print the correct response and the error as below. and i confirm the actual request and response are correct on CAN BUS.

> raise UnexpectedResponseException(response, "Server returned values for data identifier 0x%04x that was not requested and no Codec was defined for it. Parsing must be stopped." % (e.key))
udsoncan.exceptions.UnexpectedResponseException: ReadDataByIdentifier service execution returned a valid response but unexpected. Details : Server returned values for data identifier 0x3933 that was not requested and no Codec was defined for it. Parsing must be stopped. 

thank you for your help.
the code as below :
`
from can.interfaces.vector import VectorBus
from udsoncan.connections import PythonIsoTpConnection
from udsoncan.client import Client
import isotp
import udsoncan
import struct

class MyCustomCodecThatShiftBy4(udsoncan.DidCodec):
    def encode(self, val):
        val = (val << 4) & 0xFFFFFFFF
        return struct.pack('<L', val)

    def decode(self, payload):
        val = struct.unpack('<L', payload)[0]
        return val >> 4

    def __len__(self):
        return 4

isotp_params = {
    'stmin': 32,
    'blocksize': 8,
    'wftmax': 0,
    'll_data_length': 8,
    'tx_padding': 0,
    'rx_flowcontrol_timeout': 1000,
    'rx_consecutive_frame_timeout': 1000,
    'squash_stmin_requirement': False
}

bus = VectorBus(bustype='vector', app_name='CANoe', channel=[0])
tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x123, rxid=0x456)
stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)
conn = PythonIsoTpConnection(stack)
config = dict(udsoncan.configs.default_client_config)
config['data_identifiers'] = {
    0xF190: udsoncan.AsciiCodec(15)}  # Codec that read ASCII string. We must tell the length of }
with Client(conn, request_timeout=2, config=config) as client:
    response = client.read_data_by_identifier(0xF190)
    print(response.service_data.values[0xF190])
    vin = client.read_data_by_identifier_first(0xF190)
    print(vin)

`

CSG2019, 2020-09-20:
Anyone can help on this?
or could you please give me an example to read and print the 22 F1 90 with Vector tool?
thank you

pylessard, 2020-09-20:
Hi,
Sorry, I forgot that issue.  The error message is quite clear about the issue. Your server sent data for a DID that you did not request and since the is no Codec defined for that specific DID, it is impossible for the parser to continue parsing.

I wouldn't expect you to get a positive answer from your server or a valid data coming back from the UDS client since you just copy/pasted an example code and kept the demonstration codec as is... it's called ``MyCustomCodecThatShiftByFour``, I made that up, this won't work on any real device.

If you want further assistance, share a raw CAN log

Regards

CSG2019, 2020-09-20:
Hi,
thank you for your assistance and reply.
yes, this is my first time using this library, so i didn't understand well. sorry.
actually, I just try to get and print the DIDF194 response, and that need to translate from hex to ascii format.
below is the code and raw CAN log.
Do I need to define an addition function for DID F190 response?
Could you please guide me how to complete this code?
thank you. 
 
[image]

[image]

pylessard, 2020-09-20:
Hi, 
You're very close to make it work. Looking at your CAN log, you receive 17 bytes of data for your DID. Since you defined that it is a 15 bytes long payload, the parser believe that the two extra bytes are an additional DID receied by the server.  That is why it complains that it received an extra DID and doesn't know what to do with it. You have to be strict with DID length, they are not standardized, therefore manufacturer specific, and since no length information is transmitted over CAN, you need prior knowledge of your data length.

Just put 17 instead of 15 in your AsciiCodec and I believe you'll make it work.

Last thing, you mentioned DID F194, but everything seems about F190, so I assumed you made a typo.

Regards.

CSG2019, 2020-09-20:
Hi pylessard,
Thank you so much for your guidance. 
Yes, the DID length is 0x14, so it should be 17 bytes. I didn't understand the example well, now I can get and print the result I want.
You are right, I made a typo, I mean DID F190, not F194.
This library is very useful to me.
I think this topic can be closed.
thank you again.

closed 2020-09-20 by pylessard