pylessard/python-udsoncan #69: How to get the Reading DID Response by my custom formula

CSG2019, 2020-12-12:
Hi Friends,
I want to use this library to get the Reading DID response by my custom formula.
for example, I want to send 0x03 0x22 0x01 0x9E, and the response will be like: 0x05 0x62 0x01 0x9E 0x03 0x1F, and the formula is to convert 0x03 and 0x1F from hex to dec then multiply by 0.1544, so I want to get the result is 799(0x031F)*0.01544=12.33656.
I have checked the official example, but I still don't understand well. so sorry...
I don't know how to define the detail config, If you know about this, could you please help me on this. Thank you very much.
below is my original code, but it can't be run correctly...

`from can.interfaces.vector import VectorBus
from udsoncan.connections import PythonIsoTpConnection
from udsoncan.client import Client
import isotp
import udsoncan
import struct

class MyCustomCode:
    def encode(byte_1, byte_2):
        cal_value = (((byte_1 << 8) + byte_2) * 0.01544)
        return cal_value

isotp_params = {
    'stmin': 32,
    'blocksize': 8,
    'wftmax': 0,
    'll_data_length': 8,
    'tx_padding': 0,
    'rx_flowcontrol_timeout': 1000,
    'rx_consecutive_frame_timeout': 1000,
    'squash_stmin_requirement': False
}
bus = VectorBus(bustype='vector', app_name='CANalyzer', channel=[0])
ecu_address = isotp.Address(isotp.AddressingMode.Normal_29bits, txid=0x18DA10F1, rxid=0x18DAF110)
stack = isotp.CanStack(bus=bus, address=ecu_address, params=isotp_params)
conn = PythonIsoTpConnection(stack)
config = dict(udsoncan.configs.default_client_config)
config['data_identifiers'] = {0x019E: MyCustomCode.encode()}
with Client(conn, request_timeout=2, config=config) as client:
    ecu_did_response = client.read_data_by_identifier_first(0x019E)
print(ecu_did_response) `

pylessard, 2020-12-14:
Hi, you are not so far from it.  In your case, you need to define a codec with a ``decode`` method not encode. The payload comes in as a binary string. Something like that should do (untested).

    class MyCustomCodec:
        def decode(self, payload):
            return (int(payload[0]) << 8 + int(payload(1))) * 0.01544

        def __len__(self):
            return 2

config['data_identifiers'] = {0x019E: MyCustomCodec}

CSG2019, 2020-12-15:
Hello pylessard,
Thank you for your guidance.
It seems that I have to define a Class first, which contains the decode(formula) and the response length, then I can call this through the 'config'.
But I don't understand why it return (int(payload[0]) << 8 + int(payload(1))) * 0.01544, actually, I want to do something for the byte4 and byte5 of the response(0x05 0x62 0x01 0x9E 0x03 0x1F). Does the payload[0] and payload[1] means byte4 and byte5?
Could you please help to explain again? 
I tried to run below codes, but the error is TypeError: 'bytes' object is not callable
```
class MyCustomCodec(udsoncan.DidCodec):
    def decode(self, payload):
        return (int(payload[0]) << 8 + int(payload(1))) * 0.01544

    def __len__(self):
        return 2

config['data_identifiers'] = {0x019E: MyCustomCodec}
```

pylessard, 2020-12-15:
Hi,
The problem is because of the paranthesis used for byte 1. Use square bracket instead. My apologies, I wrote paranthesis in my answer (I work a lot with Matlab, and that software use paranthesis instead of square brackets). I updated my answer.

As for the byte offset, the library strip off the rest of the response. It only feeds the Did data to the Codec.

Regards

CSG2019, 2020-12-16:
Hello pylessard,

Thanks for your help. Now it can work as I hope. :)  Really appreciated.
I am new to Python and this library. I also didn't notice the parentheses issue previously. 
By the way, I can find the ``` encode ```  method in the example of ReadDataByIdentifier, I want to ask, in which case need to define this method?

pylessard, 2020-12-16:
Good,
Encode is when you use WriteDataByIdentifier

CSG2019, 2020-12-17:
Understand, thanks :)

closed 2020-12-17 by pylessard