pylessard/python-udsoncan #71: Configure issue : stmin

Kallappambk, 2020-12-18:
hello,
I am trying to configure the parameter stmin to 1miliseconds for my connection. but getting the following error.

connection : 
my_connection = IsoTPSocketConnection('vcan0', rxid=0x74F, txid=0x74E,stmin =1000)

Error : 
   Could not import vxlapi: module 'ctypes' has no attribute 'windll'
Let me know whether I am doing it correctly or is there any way to configure it.
so kindly help here, I am new to these things.
thanks in advance.

pylessard, 2020-12-29:
Hi, Sorry for the delay in the response.
Where did you see that you can pass a parameter called "stmin" to the Connection object? Successive arguments are passed to the socket ``bind`` method which expect an address only.  If you want to configure the stmin parameter, you need to use the ``socket.set_fc_opts`` method which needs to be set before binding the address.  

In your case, you will ne to make your socket first with a ``istop.socket`` object, then set stmin and finally give the socket to the connection object.

Regards

pylessard, 2021-02-07:
Can I close this issue?

closed 2021-02-15 by pylessard