pylessard/python-udsoncan #75: J2534 interface/windows based support

joeFischetti, 2021-02-24:
Hi there,
We've been using udsoncan for VW reflashing via a raspberry pi/socketcan and it has been fantastic!

I'm currently working on a J2534/dll wrapper so we can do the same from windows with a J2534 compliant interface (i.e. tactrix cable).  It's early but much of the core functionality seems to be almost there.

I've written it using udsoncan.connections BasicConnection so that it can be plugged in.  Is it something you would be interested in pulling in to the project?

Rough details are in my github repo:
https://github.com/joeFischetti/python-j2534

connections.py is modeled after the udsoncan connections.py
j2534.py is the main class and enums used by the library to connect to the interface
test_connection.py is a simple script that just establishes a connection, and sends/receives some data, then closes.

pylessard, 2021-02-24:
Hi,
I have to learn about J2534 first, but it sound very interesting. I'll revert back to you soon as as soon as I have done some readings.

Thank you!

joeFischetti, 2021-02-24:
Sounds good!
In short, it's a standard for a windows based interface that revolves around the use of a proprietary DLL. "Tactrix" is a US based company that makes a consumer grade interface (available for ~$170).  Their DLL appears to be fully compliant.  Tactrix has been used by Subaru's/Mitsubishis for tuning since the early 2000's (I believe?).  They sell the hardware and have their own free software for flashing.

My plugin takes the DLL location as a parameter and sets up an ISO-15765 connection. 

The two links below have been invaluable.
http://www.drewtech.com/support/passthru.html
https://www.tactrix.com/index.php?option=com_content&view=category&id=41:j2534&Itemid=41&layout=default

pylessard, 2021-04-21:
I merged your pull request. Thanks you

closed 2021-04-21 by pylessard