pylessard/python-udsoncan #78: Add support for $2C (Define dynamic identifier)

joeFischetti, 2021-02-27:
I'm currently using $2C manually using raw send/receive to create a dynamic identifier for memory reads.
It's fine and functional, but including $2C would clean it up.  

For example:
I want the data in the following memory locations:
d0081234 - word
d0084321 - byte

I want to access them as a single DID - F2 00

Send 2C 03 F2 00 - this clears out the existing F2 00 identifier
Send 2C 02 F2 00 14 d0 08 12 34 02 d0 08 43 21 01 - This creates F2 00 with those two memory addresses and their lengths

Now to read it, I can query using a standard 22:
22 F2 00

And the response might look like:
62 F2 00 AB CD EF - where AB CD is 0xd0081234  and EF is 0xd0084321 

It takes some setup, but it's so much nicer when attempting to read 30, 40, 50+ parameters with a high sample rate.

I have to assume I could build a codec for the identifier that parses the data back out during the decode, so that just leaves the initial setup (clear and define) as possible enhancements?

pylessard, 2021-04-21:
Hi @joeFischetti I think I missed this issue when you posted it. Just found out about it.
I haven't looked into this service, but maybe I should.  I put my time on something else these days, but Pull Requests are welcome.
Otherwise, I will eventually look into this.

Regards

pylessard, 2021-08-06:
Hi @joeFischetti , just to let you know that i've started implementation of this service.  It really is easier than I expected, it should come quite fast.

Regards!

pylessard, 2021-08-11:
Completed.
Will be released in v1.14

closed 2021-08-11 by pylessard