pylessard/python-udsoncan #81: WDBI Service Implementation

Arfaoui-Mahdi, 2021-04-29:
Hi, I'm trying to use udsoncan over PCan, I have succeeded in implementing Routine Control and RDBI services, I'm stucked now in WDBI.

Let's say that I want to execute a WDBI request with 2 params : 
1- string : ID , exemple = "0X1F8B" ... 
I managed to transform this to hex int representation with :
`int IDhex = int(ID,16)`
and pass it to the config dict to encode it to ascii(64)
`IDhex : udsoncan.AsciiCodec(64)`
so it could be acceptable by udsoncan.

For the Data, let's say that I have `data = "015A"`, I managed to transform this to bytes format with 
`dataBytes = bytes.fromhex(data)` so the routine control service could accept it. 

Giving what I did to make these two services work, how can I implement the WDBI service with Udsoncan.
I really need your help.

Thanks.

pylessard, 2021-04-30:
Hi,
Have you defined a codec for that Data Identifier?

When you call `write_data_by_identifier`, the data you give is given to a codec object that you can define. The goal of that codec is convert any kind of data into a bytes payload.

See an example here : https://udsoncan.readthedocs.io/en/latest/udsoncan/examples.html#reading-a-did-with-readdatabyidentifier

Arfaoui-Mahdi, 2021-05-03:
Hi, Thank you for your response, and I'm sorry for the late reply.
I have defined a codec : 

```
config = dict(udsoncan.configs.default_client_config)
   config['data_identifiers'] = {
      vrb : udsoncan.AsciiCodec(len(dat)),
   }
```

And now I'm getting a connection with the ECU, except that I always receive a NegativeResponse Exception, I don't know if that because of requests I'm sending or because of a bad config.

what do you think ?

pylessard, 2021-05-03:
You use the library correctly if you are able to get a response from your device. 
A negative response is your device that refuse request. Each device is different, so I cannot tell you what is the problem.

With a NegativeResponseexception comes an error code, this may help you understand what is wrong.
Possible that you do not have access right to use this service or you are not in the right diagnostic session. Maybe your device doesn't expect ASCII format.  Only you can tell.

Good luck

Arfaoui-Mahdi, 2021-05-03:
Thanks for your response, just one last question and i'll close the issue.
How can I detect NegativeResponseException code ?, i searched in docs, I've found a lot of other exceptions codes except NegResExc code.

pylessard, 2021-05-03:
The response is in the exception. See https://udsoncan.readthedocs.io/en/latest/udsoncan/exceptions.html#udsoncan.exceptions.NegativeResponseException

```
    except NegativeResponseException as e:
        print("%s (%d)" % (e.response.code_name, e.response.code))
```

you can also enable loggings, you will get lots of useful details from the library.  `udsoncan.setup_logging()`

Arfaoui-Mahdi, 2021-05-05:
Thanks a lot

closed 2021-05-05 by Arfaoui-Mahdi