pylessard/python-udsoncan #82: 0x36 transfer data not send all the data

sgnes, 2021-05-10:
I have run the same script on both windows 10 and linux(raspberry pi/ubuntu desktop), and the one on windows with Vector device is ok, but the one with Linux socketcan is not ok for the 0x36 service transfer data.

**the error:**

**the code**
```
   if args.can_interface == "Vector":
      bus = VectorBus(args.can_channel)
   elif args.can_interface == "PCAN":
      bus = PcanBus()  # Link Layer (CAN protocol) 
   elif args.can_interface == 'socketcan':
      bus = can.interface.Bus(channel='can0', bustype='socketcan_ctypes')
   tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=project_config.uds_req_phy_id,
                           rxid=project_config.uds_res_id)  # Network layer addressing scheme
   stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)  # Network/Transport layer (IsoTP protocol)
   stack.set_sleep_timing(0.001, 0.001)
   conn = PythonIsoTpConnection(stack)
```

**the log on linux:**
***BUSMASTER Ver 3.2.2***
***PROTOCOL CAN***
***NOTE: PLEASE DO NOT EDIT THIS DOCUMENT***
***[START LOGGING SESSION]***
***START DATE AND TIME 10:5:2021 20:30:15:451***
***HEX***
***SYSTEM MODE***
***START CHANNEL BAUD RATE***
***CHANNEL 1 - Vector - VN1610 Channel 1 SN - 24706 - 500000 bps***
***END CHANNEL BAUD RATE***
***START DATABASE FILES***
***END DATABASE FILES***
***<Time><Tx/Rx><Channel><CAN ID><Type><DLC><DataBytes>***
20:30:21:1132 Rx 1 0x71D s 8 02 10 03 00 00 00 00 00 
20:30:21:1173 Rx 1 0x71E s 8 06 50 03 00 32 01 F4 AA 
20:30:21:1211 Rx 1 0x71D s 8 02 10 02 00 00 00 00 00 
20:30:21:1273 Rx 1 0x71E s 8 06 50 02 00 32 01 F4 AA 
20:30:21:1343 Rx 1 0x71D s 8 02 27 11 00 00 00 00 00 
20:30:21:1373 Rx 1 0x71E s 8 06 67 11 55 55 55 55 AA 
20:30:21:1429 Rx 1 0x71D s 8 06 27 12 AA AA AA AA 00 
20:30:21:1473 Rx 1 0x71E s 8 02 67 12 AA AA AA AA AA 
20:30:21:1524 Rx 1 0x71D s 8 06 31 01 FF 00 01 00 00 
20:30:21:1573 Rx 1 0x71E s 8 03 7F 31 78 AA AA AA AA 
20:30:21:8273 Rx 1 0x71E s 8 05 71 01 FF 00 00 AA AA 
20:30:21:8342 Rx 1 0x71D s 8 10 09 34 00 33 00 4C 00 
20:30:21:8373 Rx 1 0x71E s 8 30 FF 00 AA AA AA AA AA 
20:30:21:8385 Rx 1 0x71D s 8 21 00 70 00 00 00 00 00 
20:30:21:8473 Rx 1 0x71E s 8 04 74 20 04 02 AA AA AA 
20:30:21:8536 Rx 1 0x71D s 8 11 02 36 01 ED CB 97 ED 
20:30:21:8573 Rx 1 0x71E s 8 30 FF 00 AA AA AA AA AA 
20:30:21:8592 Rx 1 0x71D s 8 21 7F 8F ED 7F 8F ED 7F 
20:30:21:8597 Rx 1 0x71D s 8 22 8F ED 7F 8F ED 7F 8F 
20:30:21:8601 Rx 1 0x71D s 8 23 ED EA 98 ED 7F 8F ED 
20:30:21:8606 Rx 1 0x71D s 8 24 DA 98 ED 7F 8F ED 7F 
20:30:21:8610 Rx 1 0x71D s 8 25 8F ED 8C 90 ED 7F 8F 
20:30:21:8614 Rx 1 0x71D s 8 26 ED 7F 8F ED 7F 8F ED 
20:30:21:8618 Rx 1 0x71D s 8 27 36 97 ED DC 98 ED 7F 
20:30:21:8622 Rx 1 0x71D s 8 28 8F ED 7F 8F ED 7F 8F 
20:30:21:8626 Rx 1 0x71D s 8 29 ED 7F 8F ED 7F 8F ED 
20:30:21:8630 Rx 1 0x71D s 8 2A 2C 97 ED 31 97 ED 7F 
20:30:21:8634 Rx 1 0x71D s 8 2B 8F ED 7F 8F ED 6B 95 
20:30:21:8638 Rx 1 0x71D s 8 2C ED 7F 8F ED 7F 8F ED 
20:30:21:8643 Rx 1 0x71D s 8 2D 7F 8F ED 7F 8F ED 7F 
20:30:21:8647 Rx 1 0x71D s 8 2E 8F ED 7F 8F ED 7F 8F 
20:30:21:8667 Rx 1 0x71D s 8 2F ED 7F 8F ED 7F 8F ED 
20:30:21:8671 Rx 1 0x71D s 8 20 7F 8F ED 7F 8F ED 7F 
20:30:21:8675 Rx 1 0x71D s 8 21 8F ED 7F 8F ED 7F 8F 
20:30:21:8679 Rx 1 0x71D s 8 22 ED 7F 8F ED 7F 8F ED 
20:30:21:8683 Rx 1 0x71D s 8 23 7F 8F ED 7F 8F ED 7F 
20:30:21:8686 Rx 1 0x71D s 8 24 8F ED 7F 8F ED 7F 8F 
[... 357 lines cut ...]
20:32:43:8134 Rx 1 0x71D s 8 2C 00 00 00 00 00 00 00 
20:32:43:8136 Rx 1 0x71D s 8 2D 03 01 07 00 F7 F6 FF 
20:32:43:8139 Rx 1 0x71D s 8 2E 01 00 00 07 04 07 00 
20:32:43:8141 Rx 1 0x71D s 8 2F 00 00 FF 04 00 00 0F 
20:32:43:8144 Rx 1 0x71D s 8 20 04 07 00 20 F5 FF 04 
20:32:43:8146 Rx 1 0x71D s 8 21 00 00 10 04 07 00 80 
20:32:43:8149 Rx 1 0x71D s 8 22 F8 FF 01 00 00 86 F1 
20:32:43:8151 Rx 1 0x71D s 8 23 07 00 00 00 FF 01 00 
20:32:43:8153 Rx 1 0x71D s 8 24 00 89 F1 07 00 00 00 
20:32:43:8156 Rx 1 0x71D s 8 25 FF 04 00 00 9E F1 07 
20:32:43:8158 Rx 1 0x71D s 8 26 00 08 F5 FF 0F 00 00 
20:32:43:8160 Rx 1 0x71D s 8 27 A2 F1 07 00 18 F5 FF 
20:32:43:8163 Rx 1 0x71D s 8 28 06 00 00 A3 F1 07 00 
20:32:43:8165 Rx 1 0x71D s 8 29 98 F6 FF 40 00 00 AB 
20:32:43:8168 Rx 1 0x71D s 8 2A F1 07 00 00 00 FF 08 
20:32:43:8170 Rx 1 0x71D s 8 2B 00 00 DF F1 07 00 00 
20:32:43:8172 Rx 1 0x71D s 8 2C 00 FF 01 00 00 03 02 
20:32:43:8175 Rx 1 0x71D s 8 2D 04 00 00 00 00 00 00 
20:32:43:8177 Rx 1 0x71D s 8 2E 00 02 02 02 00 00 00 
20:32:43:8180 Rx 1 0x71D s 8 2F 00 04 02 00 00 FF 02 
20:32:43:8182 Rx 1 0x71D s 8 20 00 00 00 00 02 02 00 
20:32:43:8185 Rx 1 0x71D s 8 21 01 FF 02 00 00 00 00 
20:32:43:8187 Rx 1 0x71D s 8 22 00 02 00 10 11 22 27 
20:32:43:8189 Rx 1 0x71D s 8 23 28 2E 31 34 36 37 3E 
20:32:43:8192 Rx 1 0x71D s 8 24 85 01 02 03 00 11 12

pylessard, 2021-05-10:
Hi,
Do you use the same version of the library on Linux and Windows.
Also, do you ignore the server timing when the issue happens? (config['use_server_timing']

I find it strange as the exception raises but clearly, the timer shouldn't expire.

sgnes, 2021-05-10:
should be the same, at least for isotp and python udsoncan, the point is not the timeout issue, we should figure out why the download data did not send all the data, right?

pylessard, 2021-05-11:
Ha!
Try increasing your network queue length
https://stackoverflow.com/questions/40424433/write-no-buffer-space-available-socket-can-linux-can

pylessard, 2021-05-11:
No problems. 
I will see if there's a way to bubble the error up to the application correctly.

Also, if you are on linux, I think you should consider looking at the IsoTP kernel module.

shinhwap, 2021-09-24:
@sgnes 
hello, can u share your isotp_params in windows 10?
I run my script in windows10 and faced the same error like you in linux.
here is my isotp_param
```
        self.isotp_params = {
        'stmin' : 32,                          # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
        'blocksize' : 8,                       # Request the sender to send 8 consecutives frames before sending a new flow control message
        'wftmax' : 0,                          # Number of wait frame allowed before triggering an error
        'll_data_length' : 8,                  # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
        'tx_padding' : 0,                      # Will pad all transmitted CAN messages with byte 0x00. None means no padding
        'rx_flowcontrol_timeout' : 1000,       # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
        'rx_consecutive_frame_timeout' : 1000, # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
        'squash_stmin_requirement' : True     # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
        }
        self.interface = interface
        self.bus = ""                                                # Link Layer (CAN protocol)
        self.tp_addr = ""                                            # Network layer addressing scheme
        self.stack = ""                                              # Network/Transport layer (IsoTP protocol)
        self.conn = ""                                               # interface between Application and Transport layer
        self.config = dict(udsoncan.configs.default_client_config)
        self.config['data_identifiers'] = {}
        self.config['p2_server_max'] = 5
        self.config['p2_star_timeout'] = 5
        self.config['p2_timeout'] = 0.5
        self.config['standard_version'] = 2006
        self.config['use_server_timing'] = False
```

Windows CAN message log:
[image]