pylessard/python-udsoncan #83: can't use func:send in the "class connections" to send data

XXIN0, 2021-05-18:
Hi!
I want to using Udsoncan over python-can to send custom data,but it did't work,can you tell me how to send custom data,thank you!

Hardware: vector CANcaseXL
here is my code:
***********************************************************************************************
import isotp
import can
from udsoncan.connections import PythonIsoTpConnection

bus = can.interface.Bus(bustype='vector', channel='0,1', app_name='Python Can',
                        bitrate=1000000, receive_own_messages=True)

isotp_params = {
    'stmin': 32,
    'blocksize': 8,
    'wftmax': 0,
    'tx_data_length': 8,
    'tx_padding': 0,
    'rx_flowcontrol_timeout': 1000,
    'rx_consecutive_frame_timeout': 1000,
    'squash_stmin_requirement': False
}

tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x703, rxid=0x70B)
tp_layer = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)
conn = PythonIsoTpConnection(tp_layer)
conn.send(b'\x02\x02\x00\x00\x22\xC3\x00\x00')
************************************************************************************************

pylessard, 2021-05-18:
Generally, when something doesn't work, it comes with an error message. Can you share it?

I think you simply forgot to open the connection.
Call `conn.open()` before send. 

A `with` statement will also oprn it.

Regard

XXIN0, 2021-05-19:
Hi
When I run the program,it did't raised an error

I'll take your suggestion to call open() before send,but It's going to keep blocking,if I call close() after send(),the data could't sent to BUS.(it also did't raised an error)

Ps: can you give an example of how to use with statement to send custom data?
here is my code(call open() before send):
*********************
bus = can.interface.Bus(bustype='vector', channel='0,1', app_name='Python Can',
                        bitrate=1000000, receive_own_messages=True)

isotp_params = {
    'stmin': 32,
    'blocksize': 8,
    'wftmax': 0,
    'tx_data_length': 8,
    'tx_padding': 0,
    'rx_flowcontrol_timeout': 1000,
    'rx_consecutive_frame_timeout': 1000,
    'squash_stmin_requirement': False
}

tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x703, rxid=0x70B)
tp_layer = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)
conn = PythonIsoTpConnection(tp_layer)

conn.open()
conn.send(b'\x10\x02')
# conn.ckose()

pylessard, 2021-05-19:
So, does it exit silently or does it block?
If it blocks, where does it block?

It seems to me that your program exit too quickly. You can add a sleep in your program. Or use the `transmitting()` function (look at the isotp examples)

Last thing, if your goal is to send raw isotp data, you don't really need the Connection object. You can use the isotp stack directly.  Connections acts as an adapter for the UDS layer.

You can follow this example : https://can-isotp.readthedocs.io/en/latest/isotp/examples.html#basic-transmission-with-python-can

XXIN0, 2021-05-19:
1、If I don't call open() before send(),it  will exit silently

2、if I don't call close() after send(),the open() ‘Thread’ will block the program
[image]

3、I try to add a sleep() before call close() in the program,it  works fine.

I'm not very good at coding and my english is not good,Thank you for your patient answer.

closed 2021-05-19 by XXIN0