pylessard/python-udsoncan #84: Add support for ODX

ha-solutions, 2021-05-24:
Hi, would it be possible to investigate adding support for ODX, specifically ODX-D which contains DTC descriptions, Diagnostic Services, PID's, Routines, IO Control and positive/negative responses?

keelung-yang, 2022-01-24:
Me too.
It's better to add OTX support too.
But they're complicated, many works need to be done.
Maybe starting an indepent ODX/OTD parser project is better than just adding them to UDSonCAN.

pylessard, 2022-01-24:
Agreed with the independent parser.
Not much time in front of me for this unfortunately

francosax, 2022-01-25:
Hi all,
I had no such time to go deeper, but seems that this tool already exist and it will be a matter of integrate with-

This is the link to the project: https://github.com/richClubb/python-uds

Franco

raleighlittles, 2023-08-01:
@ha-solutions  Tangentially related, but how exactly is the ODX file generated/retrieved? Do you just have to use the Device Identifier (0xF19E) and read the file out as bytes? https://i.imgur.com/HXQr2FF.png

1atabey1, 2023-09-11:
> Me too.
> It's better to add OTX support too.
> But they're complicated, many works need to be done.
> Maybe starting an indepent ODX/OTD parser project is better than just adding them to UDSonCAN.

@pylessard, @keelung-yang or anyone else here who considers starting a new parser project, there are a few actively maintained libraries that are already working on providing this functionality:

The python-uds package is unfortunately no longer maintained and might generally be tough to integrate with other packages. However, https://github.com/mercedes-benz/odxtools is probably worth having a look at. It is actively being maintained and updated and already in a state where it can be productively used (e.g. together with the can-isotp or udsoncan packages)

Besides that there is also https://github.com/cantools/cantools which some of you might already be using for parsing dbc files but it also supports CDD which is functionally equivalent to ODX and can be created from ODX files

jochen-zimmer, 2023-09-15:
@1atabey1 I can just say that I'm using odxtools and came here to write about this as possible extension, but you did already. I had an issue with odxtools reading my pdx file, but at the same day, the issue got accepted, solved and integrated - so they are indeed very active. You could almost just integrate the [Database class](https://github.com/mercedes-benz/odxtools/blob/main/odxtools/database.py) in [ClientConfig](https://github.com/pylessard/python-udsoncan/blob/1b93cc3cd0e09a21d48881ba53aed257f841bb89/udsoncan/typing.py#L28) (or extract certain information from Database, like the data_identifiers Dict and fill it with that info).

pylessard, 2023-09-18:
Nice!
I'm focusing on another project right now and have little time to dedicate to this one. I would gladly accept a PR (given that it is well made :) )

Cheers!