pylessard/python-udsoncan #93: send flow control frame timeout

XXIN0, 2021-07-29:
I use PythonIsoTpConnection in udsoncan.connections to send and recive can msg,but when use it to test uds service 2E to write data, it will timeout sometimes when send flow control frame, please Look at the picture below
[image]

Hardware：vector cancasexl 
Python 3.7.9
 
here is code
class UdsLib:

    def set_can_config(self, tx_id=0x703, rx_id=0x70B):

        try:
            self.bus = can.interface.Bus(bustype='vector', channel='0,1', app_name='Python Can', bitrate=500000,
                                         receive_own_messages=True)
            self.tp_addr = isotp.Address(addressing_mode=isotp.AddressingMode.Normal_11bits, txid=tx_id, rxid=rx_id)
            tp_stack = isotp.CanStack(bus=self.bus, address=self.tp_addr, params=iso_tp_params)
            self.conn = PythonIsoTpConnection(tp_stack)
        except:
            AssertionError(sys.exc_info()[0])
        else:
            logger.info("Set Can config is ok!")

    def diagnostic_request(self, request_dat):
        request_pdu = binascii.a2b_hex(request_dat)
        if not self.conn.is_open():
            self.conn.open()
        try:
            self.conn.send(request_pdu)
        except can.CanError:
            AssertionError("Send request data fail")
        else:
            logger.info('Sending %d bytes : <%s>' % (len(request_pdu), binascii.hexlify(request_pdu)))

    def diagnostic_respond(self, respond_dat):
        expect_respond_pdu = binascii.a2b_hex(respond_dat)
        try:
            resp_pdu = self.conn.wait_frame(timeout=2)

            if resp_pdu is not None:
                while resp_pdu[0] == 0x7F and resp_pdu[2] == 0x78:
                    logger.info('Received %d bytes : <%s>' % (len(resp_pdu), binascii.hexlify(resp_pdu)))
                    resp_pdu = self.conn.wait_frame(timeout=3)
        except Exception as err:
            raise err

        if resp_pdu is not None:
            logger.info('ECU: Received %d bytes : <%s>' % (len(resp_pdu), binascii.hexlify(resp_pdu)))
            if resp_pdu.find(expect_respond_pdu) == 0:
                return True
            else:
                return binascii.b2a_hex(resp_pdu)
        return resp_pdu

    def close_can(self):
        self.conn.close()
        self.bus.stop_all_periodic_tasks()
        self.bus.shutdown()
        logger.info("close the device")

pylessard, 2021-07-29:
Hi, your server requires the client to use a 50ms timeout in DiagnosticSessionControl response which is hard to respect for a software stack running in the user space under windows.

I would suggest to ignore the server timing by setting `use_server_timing` to False.

pylessard, 2021-07-29:
Same as #80 and #87.
Is 50msec an industry standard in China?

XXIN0, 2021-07-30:
I take your suggested setting **use_server_timing** to False.but also timeout,I think It's not the 50ms timeout,because it works for more than 50ms,it does not work properly after 150ms,please Look at the picture below,

[image]

- Is 50msec an industry standard in China?
yes

pylessard, 2021-07-30:
Can you enable isotp logging and share this?
Simply enable Debug level on a logger named "isotp". Then share the output.

Your CAN log doesn't show the timeout problem, so it's hard for me to know what is good or bad in the logs.

XXIN0, 2021-07-30:
[image]
`DEBUG    Connection:connections.py:51 Sending 2 bytes : [b'1001']
DEBUG    isotp:protocol.py:478 Sending :  <701> (8)      b'0210010000000000'
DEBUG    isotp:protocol.py:471 Receiving : 1627616054.878659 <701> (8)   b'0210010000000000'
DEBUG    isotp:protocol.py:471 Receiving : 1627616054.884688 <781> (8)   b'065001003201f400'
DEBUG    Connection:connections.py:78 Received 6 bytes : [b'5001003201f4']
DEBUG    Connection:connections.py:51 Sending 3 bytes : [b'22f190']
DEBUG    isotp:protocol.py:478 Sending :  <701> (8)      b'0322f19000000000'
DEBUG    isotp:protocol.py:471 Receiving : 1627616055.008502 <701> (8)   b'0322f19000000000'
DEBUG    isotp:protocol.py:471 Receiving : 1627616055.014695 <781> (8)   b'101462f190000000'
DEBUG    isotp:protocol.py:478 Sending :  <701> (8)      b'3000320000000000'
DEBUG    isotp:protocol.py:471 Receiving : 1627616055.198704 <701> (8)   b'3000320000000000'
WARNING  isotp:protocol.py:896 Reception of CONSECUTIVE_FRAME timed out.
DEBUG    Connection:connections.py:70 No data received: [TimeoutException] - Did not receive frame IsoTP Transport layer in time (timeout=2 sec)
`

XXIN0, 2021-07-30:
DEBUG    Connection:connections.py:51 Sending 2 bytes : [b'1001']
DEBUG    isotp:protocol.py:478 Sending :  <701> (8)      b'0210010000000000'
DEBUG    isotp:protocol.py:471 Receiving : 1627616054.878659 <701> (8)   b'0210010000000000'
DEBUG    isotp:protocol.py:471 Receiving : 1627616054.884688 <781> (8)   b'065001003201f400'
DEBUG    Connection:connections.py:78 Received 6 bytes : [b'5001003201f4']
DEBUG    Connection:connections.py:51 Sending 3 bytes : [b'22f190']
DEBUG    isotp:protocol.py:478 Sending :  <701> (8)      b'0322f19000000000'
DEBUG    isotp:protocol.py:471 Receiving : 1627616055.008502 <701> (8)   b'0322f19000000000'
DEBUG    isotp:protocol.py:471 Receiving : 1627616055.014695 <781> (8)   b'101462f190000000'
DEBUG    isotp:protocol.py:478 Sending :  <701> (8)      b'3000320000000000'
DEBUG    isotp:protocol.py:471 Receiving : 1627616055.198704 <701> (8)   b'3000320000000000'
WARNING  isotp:protocol.py:896 Reception of CONSECUTIVE_FRAME timed out.
DEBUG    Connection:connections.py:70 No data received: [TimeoutException] - Did not receive frame IsoTP Transport layer in time (timeout=2 sec)

[image]

pylessard, 2021-07-30:
Hi,
Well there is indeed a missing frame in the communication, the IsoTP stack is right to complain about a timeout.
After the flow control frame, there should be an answer from your server starting with byte 0x21, but instead there's nothing for 2 seconds.

[image]

XXIN0, 2021-08-02:
Hi，
I consult  the software engineers in my company，he said it was the flow control frame send timeout(>150ms),so that the serrver didn‘t responsed, so the IsoTP stack  also timeout.
The problem is why the flow control frame timeout sometimes?
[image]

pylessard, 2021-08-05:
Well, it does not "timeout". 150 msec is a timeout value set by your organization.
The question is, why is there a delay in the transmission on your setup.  When I run the stack locally, it goes faster than this.

So there is something in your setup that cause the stack to execute slowly. What is your CAN hardware? Operating system? Hardware performance, process priority, etc . 

You should dig a bit into the code and try to identify at what point the delay is introduced.