pylessard/python-udsoncan #94: IsoTPSocketConnection and 29Bit addresses

pompushko, 2021-07-31:
Hello

Doest IsoTPSocketConnection support 29Bit addresses ?
I trying to make diagnostic connection with module which use 29Bit only

``
ValueError: txid must be smaller than 0x7FF for 11 bits identifier
``

pylessard, 2021-07-31:
Hi,
Yes, absolutely.
I must admit that this Connection is the first one that I wrote and didn't evolve well with time.

The connection can accept additional arguments. If you provide an `isotp.Address` object, it will be used correctly. Refer to the [isotp doc](https://can-isotp.readthedocs.io/en/latest/isotp/addressing.html)

By setting the addressing mode to Normal_29bits, NormalFixed_29bits, Extended_29bits, Mixed_29bits, the correct socket flag will be set. 

There is a small flaw that I need to correct.  txid and rxid are set in the address object but won't be used in the UDS connection, so you need to set them again. Here's what I suggest.

    addr = isotp.Address(isotp.AddressingMode.Normal_29bits, txid=0x111111, rxid=0x111112)
    txid = addr.get_tx_arbitraton_id()    % Extract txid to give it back to the connection.
    rxid = addr.get_rx_arbitraton_id()    % Extract rxid to give it back to the connection.
    conn = IsoTPSocketConnection('can0', txid, rxid, addr)

pompushko, 2021-08-02:
Hello

Yes, that is working fine:

```
bus = SocketcanBus(channel=channel, bustype=bustype)
tp_addr = isotp.Address(isotp.AddressingMode.Normal_29bits, txid=0x18DA62F1, rxid=0x18DAF162)
stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)
conn = PythonIsoTpConnection(stack)
with Client(conn,  request_timeout=2, config=config) as client:
```

closed 2021-08-02 by pompushko