pylessard/python-udsoncan #96: CommunicationControl issue with 0x80 control_type

pompushko, 2021-08-02:
Hello

In my case, original diagnostic software use 0x80 in control_type byte.

Any ideas?

Thank you

```
2021-08-02 22:37:21 [ERROR] UdsClient: [ValueError] : Control type must be an integer between 0x0 and 0x7F
Traceback (most recent call last):
  File "main.py", line 79, in <module>
    client.communication_control(128, 3)
  File "/usr/local/lib/python3.7/dist-packages/udsoncan/client.py", line 131, in decorated
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/udsoncan/client.py", line 728, in communication_control
    request = services.CommunicationControl.make_request(control_type, communication_type)
  File "/usr/local/lib/python3.7/dist-packages/udsoncan/services/CommunicationControl.py", line 52, in make_request
    ServiceHelper.validate_int(control_type, min=0, max=0x7F, name='Control type')
  File "/usr/local/lib/python3.7/dist-packages/udsoncan/services/__init__.py", line 117, in validate_int
    raise ValueError("%s must be an integer between 0x%X and 0x%X" % (name, min, max))
ValueError: Control type must be an integer between 0x0 and 0x7F
```

pylessard, 2021-08-05:
Hi,
Values between 0x80 and 0xFF are reserved by ISO-14229, that is the reason why I restricted the range of this.
After second thought, maybe I should remove that restriction.  

I will do so soon.

pylessard, 2021-08-06:
I made a mistake! Just like #97, you need to use suppress_positive_response.  
I am rusty a bit

commit e0e5e756 (2021-08-05): Fixed #96 - Communication control allows control type between 0x80 and 0xFF
closed 2021-08-06 by pylessard