pylessard/python-udsoncan #97: UDS-00051 types

pompushko, 2021-08-02:
In original diagnostic software exists UDS "broadcast" types:

```
UDS-00051 START Tester Present on frame 18DBFEF1 PADDED (broadcast)
start periodic: CANI29BIT - 18DBFEF1 - 3E 80

UDS-00051 STOP Tester Present on frame 18DBFEF1 PADDED (broadcast)
stop periodic: 00000012
```

```
UDS-00051 Start Diagnostic Session Extended (broadcast)
  tx: CANI29BIT - 18DBFEF1 - 10 83
  rx: CANI29BIT - not required
```

```
UDS-00051 Disable Dtc Setting (broadcast)
  tx: CANI29BIT - 18DBFEF1 - 85 82
  rx: CANI29BIT - not required

UDS-00051 Enable Dtc Setting (broadcast)
  tx: CANI29BIT - 18DBFEF1 - 85 81
  rx: CANI29BIT - not required
```

```
UDS-00051 Disable Normal Message Transmission (broadcast)
  tx: CANI29BIT - 18DBFEF1 - 28 81 03
  rx: CANI29BIT - not required

UDS-00051 Enable Normal Message Transmission (broadcast)
  tx: CANI29BIT - 18DBFEF1 - 28 80 03
  rx: CANI29BIT - not required
```

pylessard, 2021-08-02:
There's no question in your post.
I don't know what is the "original" software.
Please, put some effort on your side if you want mr to put some.

pompushko, 2021-08-02:
Hello

Sorry for that  :)

Could you please extend range in request?

pylessard, 2021-08-05:
I have no idea what's going on here.
I think your last emssage refers to #96. If that's the case, it's done and will be published in v1.14.

But as for this issue right here..  no clue. Maybe the same as #91

pompushko, 2021-08-06:
TesterPresent (0x3E)
with broadcast should be 0x80 byte

pylessard, 2021-08-06:
ah!
No need to change the library, setting the MSB to 1 means that you don't expect an answer from the server. In that case you do:

    with client.suppress_positive_response:
       client.tester_present()

This will force bit #7 of the subfunction to be 1, meaning 0x80 if you pass a subfunction of 0 (default value).

Regards

pompushko, 2021-08-06:
> ah!
> No need to change the library, setting the MSB to 1 means that you don't expect an answer from the server. In that case you do:
> 
> ```
> with client.suppress_positive_response:
>    client.tester_present()
> ```
> 
> This will force bit 7 of the subfunction to be 1, meaning 0x80 if you pass a subfunction of 0 (default value).
> 
> Regards

Thank you!

So, same for "Start Diagnostic Session Extended"?
Diagnostic app send 0x83

pylessard, 2021-08-06:
Exactly!

pompushko, 2021-08-06:
> Exactly!

Wow, Nice, let me check other routines like "Disable Normal Message Transmission"

closed 2021-08-11 by pylessard

pompushko, 2023-08-09:
Well, I facing such issue again :D

In short: I need somehow to send broadcast message. 
Its like 18DBFEF1 address for broadcast
and 18DAFEF1 with 18DAF1FE for UDS with particular ECU

Could you please help me with that? Thank you!

pylessard, 2023-08-09:
So you are using [NormalFixed addressing.  ](https://can-isotp.readthedocs.io/en/latest/isotp/addressing.html#normal-fixed-addressing) with source address F1 and target address FE

What is your isotp layer? If you use [my library](https://github.com/pylessard/python-can-isotp), you can create an address like that

```python
address = isotp.Address(istop.AddressingMode.NormalFixed_29bits, source_address = 0xF1, target_address=0xFE)
```

You ID should then be fine. Switching between functional and physical addressing with a single UDS client is a bit hacky; I have not put effort into making this cleaner. Check example below.

```
addr =  isotp.Address(istop.AddressingMode.NormalFixed_29bits, source_address = 0xF1, target_address=0xFE)
stack = isotp.CanStack(bus, address=addr)
conn = PythonIsoTpConnection(stack)  
with Client(conn) as client:
    stack.params.set('default_target_address_type ', isotp.TargetAddressType.Functional)
    client.something() // Functional
    stack.params.set('default_target_address_type ', isotp.TargetAddressType.Physical)
    client.something() // Physical
```

pompushko, 2023-08-09:
Thank you!

Well, my addresses is
```
addr =  isotp.Address(istop.AddressingMode.NormalFixed_29bits, source_address = 0x18da20f1, target_address=0x18daf120)
```
but I need send sometimes broadcast messages during connection:
```
<line>UDS-07209 Start Diagnostic Session Extended (broadcast)</line>
<line> tx: CANCH29BIT - 18dbfef1 - 10 83</line>
<line> rx: CANCH29BIT - not required</line>
<line>KWP-2000 Start Diagnostic Session Extended (broadcast)</line>
<line> tx: CANCH29BIT - 18db33f1 - 10 92</line>
<line> rx: CANCH29BIT - not required</line>
<line>UDS-07209 Start Diagnostic Session Extended (broadcast)</line>
<line> tx: CANI29BIT - 18dbfef1 - 10 83</line>
<line> rx: CANI29BIT - not required</line>
<line>KWP-2000 Start Diagnostic Session Extended (broadcast)</line>
<line> tx: CANI29BIT - 18db33f1 - 10 92</line>
<line> rx: CANI29BIT - not required</line>
<line>UDS-07209 Start Diagnostic Session Extended (broadcast)</line>
<line> tx: CANC29BIT - 18dbfef1 - 10 83</line>
<line> rx: CANC29BIT - not required</line>
<line>KWP-2000 Start Diagnostic Session Extended (broadcast)</line>
<line> tx: CANC29BIT - 18db33f1 - 10 92</line>
<line> rx: CANC29BIT - not required</line>
```

is it possible to do that during connection?

pompushko, 2023-08-09:
Thank you for help
But, still doesnt work ;(

```
  vcan0  18DA20F1   [8]  02 10 03 00 00 00 00 00
  vcan0  18DAF120   [7]  06 50 03 00 1E 0B B8
  vcan0  **18DA20F1**   [8]  02 3E 80 00 00 00 00 00
  vcan0  **18DA20F1**   [8]  02 85 82 00 00 00 00 00
  vcan0  **18DA20F1**   [8]  03 28 81 03 00 00 00 00
```

Still doesnt work...

```
    bus = can.Bus(channel=args.interface, interface='socketcan')
    ecu_address = isotp.Address(isotp.AddressingMode.NormalFixed_29bits, source_address=parsed_data.TesterAddressClient, target_address=parsed_data.ECUAddressServer)
    stack = isotp.CanStack(bus=bus, address=ecu_address, params=isotp_params)  
    conn = PythonIsoTpConnection(stack)
    with Client(conn,  request_timeout=5) as client:
        response = client.change_session(3)
        if response.code == Response.Code.PositiveResponse:
            logger_app.logger.info("UDS session changed to Extended")

        stack.params.set('default_target_address_type', isotp.TargetAddressType.Functional)
        with client.suppress_positive_response:
            client.tester_present()
            client.control_dtc_setting(ControlDTCSetting.SettingType.off)
            communication_type = udsoncan.CommunicationType(0, normal_msg=True, network_management_msg = True)
            client.communication_control(CommunicationControl.ControlType.enableRxAndDisableTx, communication_type)

        stack.params.set('default_target_address_type', isotp.TargetAddressType.Physical)
```

In my case, I should see in candump something like that
``
  vcan0  18DA20F1   [8]  02 10 03 00 00 00 00 00
  vcan0  18DAF120   [7]  06 50 03 00 1E 0B B8
  vcan0  **18DBFEF1**   [8]  02 3E 80 00 00 00 00 00
  vcan0  **18DBFEF1**   [8]  02 85 82 00 00 00 00 00
  vcan0  **18DBFEF1**   [8]  03 28 81 03 00 00 00 00
``