pylessard/python-udsoncan #98: [InvalidResponseException] : service execution returned an invalid response. Reason : Payload is empty

MasterCodeIT, 2021-08-13:
**Sending back to back request give me this error:**

`[InvalidResponseException] : service execution returned an invalid response. Reason : Payload is empty`

**So I attempted to add an exception to resend a different byte when the error is flagged:**

`            except InvalidResponseException as e:
                print('Server sent an invalid payload : %s' % e.response.original_payload)
                data = b'\x22\x20\x00'
                conn.send(data)`

**but received this:** 

`<PositiveResponse: [TesterPresent] - 1 data bytes at 0x0a43d4f0>
W1D2F3KD2JA29R791
[InvalidResponseException] : service execution returned an invalid response. Reason : Payload is empty
Server sent an invalid payload : b''
<function conn.send at 0x05EFCBF8>`

**Any ideas? Not matter if I call different request, even added an ecureset request to the "except" but still no dice. It does not matter what I ask for. Individual works well, but back to back triggers this error**

MasterCodeIT, 2021-08-13:
**full traceback below:**

`[InvalidResponseException] : service execution returned an invalid response. Reason : Payload is empty`
`Server sent an invalid payload : b''`
`[InvalidResponseException] : service execution returned an invalid response. Reason : Payload is empty`
`[InvalidResponseException] : service execution returned an invalid response. Reason : Payload is empty`
`Traceback (most recent call last):`
`  File "\requests.py", line 368, in processB`
`    client.tester_present()`
`  File "\lib\site-packages\udsoncan\client.py", line 131, in decorated`
`    return func(self, *args, **kwargs)`
`  File "\lib\site-packages\udsoncan\client.py", line 332, in tester_present`
`    response = self.send_request(req)`
`  File "\lib\site-packages\udsoncan\client.py", line 1778, in send_request`
`    raise InvalidResponseException(response)`
`udsoncan.exceptions.InvalidResponseException: service execution returned an invalid response. Reason : Payload is empty`

`During handling of the above exception, another exception occurred:`

`Traceback (most recent call last):`
`  File "\requests.py", line 380, in processB`
`    client.read_data_by_identifier_first(0xF190)`
`  File "lib\site-packages\udsoncan\client.py", line 131, in decorated`
`    return func(self, *args, **kwargs)`
`  File "lib\site-packages\udsoncan\client.py", line 358, in read_data_by_identifier_first`
`    response = self.read_data_by_identifier(didlist)`
`  File "\lib\site-packages\udsoncan\client.py", line 131, in decorated`
`    return func(self, *args, **kwargs)`
`  File "\lib\site-packages\udsoncan\client.py", line 389, in read_data_by_identifier`
`    response = self.send_request(req)`
`  File "\lib\site-packages\udsoncan\client.py", line 1778, in send_request`
`    raise InvalidResponseException(response)`
`udsoncan.exceptions.InvalidResponseException: service execution returned an invalid response. Reason : Payload is empty`

`Process finished with exit code -1073740791 (0xC0000409)`

pylessard, 2021-08-13:
Hi,
Can you share a CAN log?
What Connection do you use?

MasterCodeIT, 2021-08-13:
> Hi,
> Can you share a CAN log?
**I sniff it to get it.** 

> What Connection do you use?

**J2534 connection**

MasterCodeIT, 2021-08-13:
It is very odd, the minute I initiate a request, i can ask for it and it responds positive, but then if I try to request again, I get the error stating the payload is empty, even with the same request

MasterCodeIT, 2021-08-13:
**Click on ID button:** 
`<PositiveResponse: [TesterPresent] - 1 data bytes at 0x0a08ee50>`
`2021-08-13 17:33:06 [INFO] UdsClient: ReadDataByIdentifier<0x22> - Reading data identifier : 0xf190 (VINDataIdentifier)`
`2021-08-13 17:33:06 [DEBUG] Connection: Sending 3 bytes : [b'22f190']`
`2021-08-13 17:33:06 [DEBUG] Connection: Received 20 bytes : [b'62f1905744445a46384b42324a41323933373932']`
`2021-08-13 17:33:06 [INFO] UdsClient: Received positive response for service ReadDataByIdentifier (0x22) from server.`
`W1D2F3KD2JA29R791`
`2021-08-13 17:33:06 [INFO] Connection: J2534 Connection closed`

Wait 1 min, **Click on ID button** again (2nd time)
`processB initiated`
`2021-08-13 17:33:08 [INFO] Connection: J2534 Connection opened`
`2021-08-13 17:33:08 [INFO] UdsClient: TesterPresent<0x3e> - Sending TesterPresent request`
`2021-08-13 17:33:09 [DEBUG] Connection: Sending 2 bytes : [b'3e00']`
`2021-08-13 17:33:09 [DEBUG] Connection: Received 0 bytes : [b'']`
`2021-08-13 17:33:09 [ERROR] UdsClient: [InvalidResponseException] : service execution returned an invalid response. Reason : Payload is empty`
`Server sent an invalid payload : b''`
`2021-08-13 17:33:09 [INFO] Connection: J2534 Connection closed`

MasterCodeIT, 2021-08-13:
Even sending the bytes raw:

**Hit button 1st time:**
`2021-08-13 17:59:08 [INFO] Connection: J2534 Connection opened`
`2021-08-13 17:59:08 [DEBUG] Connection: Sending 3 bytes : [b'22f190']`
`2021-08-13 17:59:08 [DEBUG] Connection: Received 20 bytes : [b'62f1905744445a46384b42324a41323933373932']`
`b'b\xf1\x90W1D2F3KD2JA29R791'`
`2021-08-13 17:59:08 [INFO] Connection: J2534 Connection closed`

**Hit button 2nd time:**
`processB initiated`
`2021-08-13 17:59:12 [INFO] Connection: J2534 Connection opened`
`2021-08-13 17:59:12 [DEBUG] Connection: Sending 3 bytes : [b'22f190']`
`2021-08-13 17:59:12 [DEBUG] Connection: Received 0 bytes : [b'']`
`b''`
`2021-08-13 17:59:12 [INFO] Connection: J2534 Connection closed`

**Hit button 3rd time:**
`processB initiated`
`2021-08-13 17:59:15 [INFO] Connection: J2534 Connection opened`
`2021-08-13 17:59:15 [DEBUG] Connection: Sending 3 bytes : [b'22f190']`
`2021-08-13 17:59:15 [DEBUG] Connection: Received 0 bytes : [b'']`
`b''`
`2021-08-13 17:59:15 [INFO] Connection: J2534 Connection closed`

**Hit button 4th time:**
`processB initiated`
`2021-08-13 17:59:17 [INFO] Connection: J2534 Connection opened`
`2021-08-13 17:59:17 [DEBUG] Connection: Sending 3 bytes : [b'22f190']`
`2021-08-13 17:59:17 [DEBUG] Connection: Received 0 bytes : [b'']`
`b''`
`2021-08-13 17:59:17 [INFO] Connection: J2534 Connection closed`

pylessard, 2021-08-15:
Hi,
I can't help much unless you provide a CAN log. They usually are easy to get as most drivers will handle multiple subscribers. No need to add additional hardware, just a software that can read the CAN hardware input.

closed 2021-12-28 by pylessard