pylessard/python-udsoncan #102: How to increase delay between 0x36 transfer

shinhwap, 2021-09-24:
Hello,

I'm using python-udsoncan on Windows 10.
I've testing transfer data with below isotp params and work good.
```
        self.isotp_params = {
        'stmin' : 32,                          # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
        'blocksize' : 8,                       # Request the sender to send 8 consecutives frames before sending a new flow control message
        'wftmax' : 0,                          # Number of wait frame allowed before triggering an error
        'll_data_length' : 8,                  # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
        'tx_padding' : 0,                      # Will pad all transmitted CAN messages with byte 0x00. None means no padding
        'rx_flowcontrol_timeout' : 5000,       # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
        'rx_consecutive_frame_timeout' : 5000, # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
        'squash_stmin_requirement' : False     # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
        }
        self.config = dict(udsoncan.configs.default_client_config)
        self.config['data_identifiers'] = {0xF184: MyCustomCodec}
        self.config['p2_star_timeout'] = 60
        self.config['p2_timeout'] = 10
        self.config['use_server_timing'] = False
```
Log:
[image]

But we want to speed up the transfer time
So according to isotp module 
we set squash_stmin_requirement to True for windows system
But it will send too fast between CF
[image]
Our uds software cannot handle this speed
Is there any solution to solve this problem?

shinhwap, 2021-09-24:
Currently I wrote a delay function in source code and put into transmit_CF function
```
    def sleep(self, duration, get_now=time.perf_counter):
        now = get_now()
        end = now + duration
        while now < end:
            now = get_now()
```
```
        elif self.tx_state == self.TxState.TRANSMIT_CF:
            if self.timer_tx_stmin.is_timed_out() or self.params.squash_stmin_requirement:
                data_length = self.params.tx_data_length-1-len(self.address.tx_payload_prefix)
                msg_data = self.address.tx_payload_prefix + bytearray([0x20 | self.tx_seqnum]) + self.tx_buffer[:data_length]
                arbitration_id 	= self.address.get_tx_arbitraton_id()
                output_msg = self.make_tx_msg(arbitration_id, msg_data)
                self.tx_buffer = self.tx_buffer[data_length:]
                self.tx_seqnum = (self.tx_seqnum + 1 ) & 0xF
                self.timer_tx_stmin.start()
                self.tx_block_counter+=1
                self.sleep(0.001)
```
Now the transfer data will delay 1ms between each data
[image]

If there is another solution, please tell me.
Thank you.

pylessard, 2021-09-25:
hi @shinhwap,
Unfortunately, Windows is a bit limitative here. You can have a look at this thread : https://github.com/pylessard/python-can-isotp/issues/48

Maybe if you share some information specific to your implementation, I could help more. 

- Bitrate used
- Fastest speed your server can tolerate
- Size of the payload
- Length it takes, length you'd like to have
- Size of data chunks (I see 128 bytes)
- etc.

shinhwap, 2021-09-27:
hi @pylessard,
I'm using  (Vector interface, windows 10)  and 500KB can bus
```
self.bus = can.interface.Bus(bustype='vector', channel=0, bitrate=500000, app_name="xlCANcontrol", 
                                     data_bitrate=2000000)
self.tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=requeseID, rxid=responseID)
self.stack = isotp.CanStack(bus=self.bus, address=self.tp_addr,
                                    params=self.isotp_params)  # Network/Transport layer (IsoTP protocol)
 self.stack.set_sleep_timing(0.001, 0.001)
self.conn = PythonIsoTpConnection(self.stack)  # interface between Application and Transport layer
```
data
[image]
data chuck: 128 bytes

I'm not sure how fast can our server tolerate
But according this log
[image]
we will stuck at end of a data chuck transfer

According this issues
https://github.com/pylessard/python-can-isotp/issues/48
It suggest using CANFD but we do not support

It's OK with this solution
https://github.com/pylessard/python-udsoncan/issues/102#issuecomment-926384367
It will take about 108~120 sec for complete flash procedure
We can accept that time

I was just curious about why If I set squash_stmin_requirement to True
It will send 4 messages without any transmission times between CFs but the above issue didn't have this problem

pylessard, 2021-09-27:
A quick improvement would be to increase the chunk size, effetively reducing the number of sleep() call, saving 16ms each time.

It is the expected behaviour of `squash_stmin_requirement` to send everything as fast as possible. I am not sure what you mean by "the above issue", but if you talk about pylessard/python-can-isotp#48, looking at the can log, I see 4 messages per millisecs, which is what you would get.

Regards

shinhwap, 2021-09-28:
OK,
Since our server limit the chunk size 128 byte per CF.
I think there will not be a more efficient way for improvement.
You can close this issue now.
Thank you for your reply!

closed 2021-09-29 by shinhwap