pylessard/python-udsoncan #105: No BRS when using FD

catosh, 2021-12-14:
Hi, 
setting up the bus interface for fd with BRS does not set the BRS bit and does not change the arbitration speed. 

Here a minimum example:
```
RX_CHANNEL = 0x1C4217D8
TX_CHANNEL = 0x1C4017D8

isotp_params = {
   'stmin': 20,                          # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
   'blocksize': 64,                      # Request the sender to send 8 consecutives frames before sending a new flow control message
   'wftmax': 0,                          # Number of wait frame allowed before triggering an error
   'tx_data_length': 8,                 # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
   'tx_data_min_length': None,         # Minimum length of CAN messages. When different from None, messages are padded to meet this length. Works with CAN 2.0 and CAN FD.
   'tx_padding': 0,                      # Will pad all transmitted CAN messages with byte 0x00.
   'rx_flowcontrol_timeout': 1000,       # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
   'rx_consecutive_frame_timeout': 1000, # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
   'squash_stmin_requirement': False,    # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
   'max_frame_size': 4095               # Limit the size of receive frame.
}

bus = can.interface.Bus(bustype='vector', app_name='testApp', channel=2, bitrate=500000, fd=True, data_bitrate=1000000)
tp_addr = isotp.Address(isotp.AddressingMode.Normal_29bits, rxid=RX_CHANNEL, txid=TX_CHANNEL)
stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params) # Network/Transport layer (IsoTP protocol)
conn = PythonIsoTpConnection(stack)

with Client(conn, request_timeout=1) as client:                                     # Application layer (UDS protocol)
   client.change_session(1)
```

On the Oscilloscope I can see that no BRS bit is set and no baud rate switch is performed.

catosh, 2021-12-14:
Addendum: 
the same issue is present when I configure correctly the isotp_parameters adding 
``
 'can_fd': True

pylessard, 2021-12-14:
Hi, try this isotp parameter. The list of parameters in the example is outdated.

https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#bitrate_switch

catosh, 2021-12-14:
Hi pylessard, 
I was using version 1.6.
using version 1.7 with bitrate_switch everything works fine. 

Sorry for bothering, I try to avoid updating packages if not strictly needed.

closed 2021-12-14 by catosh

pylessard, 2021-12-14:
Alright,
I always try to keep backward compatibility, so updating shouldn't be a problem (in theory :) )

Good luck

catosh, 2021-12-14:
As a side question, does the version 1.7 handle tx_data_length bigger than 8 bytes? I am using FD where the ECU expects up to 64 bytes payloads.