pylessard/python-udsoncan #107: Question about timeout

kadirtunc, 2021-12-25:
Hi. There is an nopen issue about flashing bootloader. I have two devices. And I could flash bootloader one of them. But other device is not responding when I want to unlock security access. There is my codes for unlock security access.

My first device timeout is about 900ms but it is working perfectly. But other device want to get 30 xx xx message in 10-20ms.

Here is my codes.
```
from can.interfaces.pcan import PcanBus
from udsoncan.connections import PythonIsoTpConnection
from udsoncan.services import *
from udsoncan.client import Client
from udsoncan import *
import isotp
import time
from Crypto.Cipher import AES

bus = PcanBus()                                          # Link Layer (CAN protocol)
tp_addr = isotp.Address(isotp.AddressingMode.Normal_29bits, txid=0x18DA5535, rxid=0x18DA3555) # Network layer addressing scheme
stack = isotp.CanStack(bus=bus, address=tp_addr)               # Network/Transport layer (IsoTP protocol)
conn = PythonIsoTpConnection(stack)                                                 # interface between Application and Transport layer

def myalgo(level, seed, params):
   cryptoHandle = AES.new(bytes([0x0,0x1,0x2,0x3,0x4,0x5,0x6,0x7,0x8,0x9,0xA,0xB,0xC,0xD,0xE,0xF]),AES.MODE_ECB)
   return cryptoHandle.encrypt(bytes(seed))

with Client(conn, request_timeout=10) as client:                                     # Application layer (UDS protocol)
    client.change_session(DiagnosticSessionControl.Session.extendedDiagnosticSession)
    client.communication_control(3,3)
    client.change_session(DiagnosticSessionControl.Session.programmingSession)
    client.config['security_algo'] = myalgo
    client.unlock_security_access(1)

```

When I want to send uds commands manually and correct times, it works properly. And I can get seed.

Here is my question. How can I set mt tx message more faster? And where is the 30 xx xx message is transmitted? 

isotp_params = {
   'stmin' : 10,                          # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
   'blocksize' : 10,                       # Request the sender to send 8 consecutives frames before sending a new flow control message
   'wftmax' : 0,                          # Number of wait frame allowed before triggering an error
   'tx_data_length' : 8,                 # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
   'tx_data_min_length' : 3,         # Minimum length of CAN messages. When different from None, messages are padded to meet this length. Works with CAN 2.0 and CAN FD.
   'tx_padding' : 0,                      # Will pad all transmitted CAN messages with byte 0x00.
   'rx_flowcontrol_timeout' : 1000,       # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
   'rx_consecutive_frame_timeout' : 1000, # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
   'squash_stmin_requirement' : True,    # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
   'max_frame_size' : 4095               # Limit the size of receive frame.
}

I used this isotp params but It did not work. 
Thanks .

pylessard, 2021-12-27:
Hi sending the flow control frame within 20ms will be hard.
The flow control is sent in the isotp module which is handled by the connection object. The Connection object calls the `sleep` method between calls to stack.process. That is where your issue is.  On windows, when calling `sleep`, it may take 16ms for the process to be reactivated.

You can try to set smaller sleep time on you stak with `set_sleep_timing` : 
https://github.com/pylessard/python-can-isotp/blob/cde59f4a9e0c1b09064e3a496adb9eeddc0a6f87/isotp/protocol.py#L674

Also, some other people tried to improve this, if you look at the clsoed issue, you may find something.  I think someone even implemented a spinlock mehcanism to avoid calling `sleep`. it does make the CPU go 100% during that time though.

Best regards

kadirtunc, 2021-12-27:
https://github.com/pylessard/python-can-isotp/blob/cde59f4a9e0c1b09064e3a496adb9eeddc0a6f87/isotp/protocol.py#L386

I solved my problem with change this line to 0.2

closed 2021-12-27 by kadirtunc