pylessard/python-udsoncan #108: Issue in reading the data using ReadDatabyIdentifier

sercee, 2022-01-05:
Hello,

I want to use udsoncan library to get the DID response,for example, I want to send 0x03 0x22 0x2A 0x18, and the response will be like: 0x05 0x62 0x2A 0x18 0x01 0x10, but with the below code im getting the diffrent rensponce(i want the complete responce but its giving only 2 bytes of data in responce).

I don't know how to define the detail config and i dont want to use MyCustomCodecThatShiftBy4 function, If you know about this, could you please help me on this. code and the responce detalsare attached below.

Thank you.

from can.interfaces.vector import VectorBus
from udsoncan.connections import PythonIsoTpConnection
from udsoncan.client import Client
import isotp
import struct

import udsoncan
from udsoncan.connections import IsoTPSocketConnection
from udsoncan.client import Client
import udsoncan.configs

isotp_params = {
   'stmin' : 32,                          # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
   'blocksize' : 8,                       # Request the sender to send 8 consecutives frames before sending a new flow control message
   'wftmax' : 0,                          # Number of wait frame allowed before triggering an error
   'll_data_length' : 8,                  # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
   'tx_padding' : 0,                      # Will pad all transmitted CAN messages with byte 0x00. None means no padding
   'rx_flowcontrol_timeout' : 1000,       # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
   'rx_consecutive_frame_timeout' : 1000, # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
   'squash_stmin_requirement' : False     # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
}

class MyCustomCodecThatShiftBy4(udsoncan.DidCodec):
   def encode(self, val):
      val = (val << 4) & 0xFFFFFFFF # Do some stuff
      return struct.pack('<L', val) # Little endian, 32 bit value

   def decode(self, payload):
      val = struct.unpack('<L', payload)[0]  # decode the 32 bits value
      return val >> 4                        # Do some stuff (reversed)

   def __len__(self):
      return 4    # encoded paylaod is 4 byte long.

# config = dict(udsoncan.configs.default_client_config)
client_config = {
'exception_on_negative_response'	: False,
	'exception_on_invalid_response'		: False,
	'exception_on_unexpected_response'	: False,
	'security_algo'				: None,
	'security_algo_params'		: None,
	'tolerate_zero_padding' 	: True,
	'ignore_all_zero_dtc' 		: True,
	'dtc_snapshot_did_size' 	: 2,		# Not specified in standard. 2 bytes matches other services format.
	'server_address_format'		: None,		# 8,16,24,32,40
	'server_memorysize_format'	: None,		# 8,16,24,32,40
	'data_identifiers' 			: {0x2A81:MyCustomCodecThatShiftBy4,
	'input_output' 				: {},
    'p2_timeout' :1,
    'p2_server_max':5,
    'request_timeout':5,
    'p2_star_timeout':5
}}

bus = VectorBus(channel=0, bitrate=500000)                                          # Link Layer (CAN protocol)
tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x788, rxid=0x789) # Network layer addressing scheme
# print(tp_addr)
stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)               # Network/Transport layer (IsoTP protocol)
# print(stack)
conn = PythonIsoTpConnection(stack)
with Client(conn,  request_timeout=2, config=client_config) as client:
   response1 = client.read_data_by_identifier(0x2A81)
   print("Response: ",response1.data) # This is a dict of DID:Value

[image]

sercee, 2022-01-23:
Hi @pylessard, please suggest the solution

pylessard, 2022-01-23:
Hi,
Your question is very hard to understand. First of all, please format your code so it can be readable.

For each DID that you have, you need to define a codec that converts the data into bytes (encode) and bytes into data (decode). You also have to specify the expected length of the binary paylaod (len).  Then you assign that codec to your DID and it should work.

If you want more specific help, please provide a CAN log and also a readable piece of code.

Regards