pylessard/python-udsoncan #109: flexray

danielhrisca, 2022-01-18:
Hello @pylessard 

I'm working on integrating Flexray support for UDS. 
This would need some minor changes to this packages and bigger changes on ``python-can-isotp``:
* new Address, PDU, TransportLayer and Stack classes for Flexray

Do you think this changes could work? Or do you think a separate package for Flexray isotp would be better?

pylessard, 2022-01-22:
Hi @danielhrisca,
I'm am not so familiar with flexray, but shouldn't it replace the Link Layer, meaning replace the pythoncan module? 
Is there a standard that explain how UDS over flexray should be done? Does it use ISOTP as well?

In any case, if the modification are major and interfaces are to be broken, I would maybe prioritize creating a new module for flexray.

danielhrisca, 2022-01-22:
Yes it is mostly the transport layer that needs adjusting.

In this package it is needed to remove this hard assert if other transports are to be supported https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/connections.py#L448 with the PythonIsoToConnection class

pylessard, 2022-01-22:
Alright, if you can make something that doesn't break the interface, i'm all good.
My requirements are to be backward compatible and keeping the unit test passing (unless a test needs an update for reasonable reason).

Thanks!

pylessard, 2023-08-13:
Should I close that?

closed 2023-08-18 by pylessard